Rhoban/onshape-to-robot #7: Troubleshooting trunkOccurrence issue

marshallmassengill, 2019-07-23:
This is the error we received when running the code.  Trying to figure out what we need to troubleshoot to resolve this.  Any help provided is appreciated.  Thanks!

(env) ubuntu@ubuntu:~/onshape-to-robot$ ./onshape-to-robot.py robots/swervedrive
* Retrieving workspace ID ...
- Workspace id: 8d370ad12960758f3af7ebdf
* Retrieving elements in the document, searching for the assembly...
- Found assembly, id: f36c9e0cbc8985502f31fac7, name: "Swerve Chassis"
* Retrieving assembly
* Getting assembly features, scanning for DOFs...
Found dof: bl_angle
Found dof: bl_speed
Found dof: br_angle
Found dof: br_angle
Found dof: fr_angle
Found dof: fr_speed
Found dof: fl_angle
Found dof: fl_speed
- Found 4 DOFs
* Building robot tree
Traceback (most recent call last):
  File "./onshape-to-robot.py", line 214, in <module>
    trunkOccurrence = getOccurrence([trunk])
  File "./onshape-to-robot.py", line 97, in getOccurrence
    return occurrences[tuple(path)]
KeyError: (None,)

Gregwar, 2019-07-23:
Hello,

First, the number of DOFs doesnt match the list
Can you double check that, for each DOF, you clicked the child connector before the parent one?

What are "speed" dofs to you?

Gregwar, 2019-07-23:
If your error is not solved by the above I can have a look at the assembly

marshallmassengill, 2019-07-23:
We're working on importing a "swerve-drive" drive base. In our case, this is four wheels that are independently capable of rotating 360 degrees as well as driving.  We've broken this down into speed and angle for each wheel pod.  I did verify that each child connector is before the parent in the mates - though this is a model originally done in SolidWorks and imported into OnShape - it's likely there are other issues we don't know about.  I can add you to the assembly if that will help, can you email me your OnShape info (marshall{at}team900{dot}org).

Gregwar, 2019-07-23:
Ok having a look at your assembly, it appear that there is several parts in the assembly that should be the "main" body (the root of the robot trunk) and that are not pre-assembled

I recommend making pre-assembly of all components (parts that are moving together) with mate connectors and use them in the final assembly

You can have a look at the example here:
https://cad.onshape.com/documents/11a7f59e37f711d732274fca/w/7807518dc67487ad405722c8/e/5233c6445c575366a6cc0d50

marshallmassengill, 2019-07-23:
That makes some sense.  We'll try to resolve that and give it a go again.

Thank you for taking a look and making this tool available.

---Marshall

Gregwar, 2019-07-23:
I hope you will get it work
I am actually using this tool in my workflow now and this is convenient but there is still missing features

marshallmassengill, 2019-07-26:
Well, we got it a bit further along.  We had to switch to a different
imported model and then delete and hide a lot of components but we are now
hitting an error that we can't get past as the component that is throwing
the error is part of the mated assembly.  Here is the error:

- Adding part 110-002 - bearing holding plate <1>
Traceback (most recent call last):
  File "./onshape-to-robot.py", line 376, in <module>
    buildRobot(tree, np.matrix(np.identity(4)))
  File "./onshape-to-robot.py", line 330, in buildRobot
    addPart(occurrence, matrix)
  File "./onshape-to-robot.py", line 248, in addPart
    stl = client.part_studio_stl_m(part['documentId'],
part['documentMicroversion'], part['elementId'], part['partId'])
  File "/home/ubuntu/onshape-to-robot/onshape_api/client.py", line 271, in
part_studio_stl_m
    return self.cache_get('part_stl', (did, mid, eid, partid), invoke)
  File "/home/ubuntu/onshape-to-robot/onshape_api/client.py", line 121, in
cache_get
    f = open(fileName, 'wb')
IsADirectoryError: [Errno 21] Is a directory:
'/home/ubuntu/onshape-to-robot/onshape_api/cache/part_stl__5560e74e8c247e8dd7194eac_563c50f26487e8e2daf51ef4_014a7263dd68d18a2ab5925a_JF/'

Gregwar, 2019-07-26:
What if you run `clear-cache.sh` and start again?

The script currently tries to open a file to write the result of a STL query through API in cache but it turns out that this file is a directory (I ignore why this can happen)

Gregwar, 2019-07-26:
And also can you share the updated assembly you are now trying to import with me ?

marshallmassengill, 2019-07-26:
Just shared it with you.  We did try clearing the cache and that did not
seem to help.  I suspect this is a matter of subassemblies of subassemblies
or something similar.

---Marshall

Gregwar, 2019-07-26:
Ok, alternatively you can try bypassing the cache by changing `onshape_api/client.py` `cache_get()` method to something like:

```
    def cache_get(self, method, key, callback):
        return callback().content
```

But this is going to take a while
I can't run import on your part because it appear I don't have access to all subassemblies

Can you send me the complete output by email ?

marshallmassengill, 2019-07-26:
Yeah - let me get you added to everything too.

Gregwar, 2019-07-26:
OK apparently the part ids provided by OnShape can contains "/", which is not good
I commited a fix that uses sha1 digest hash of partid instead of directly partid so we can be totally agnostic on what it contains

Gregwar, 2019-07-26:
Ok so the good news is that your model can now load on PyBullet

I anticipate on the next issue, on very complex assemblies, even if this script will work it might be unusable in PyBullet for example because of the complexity, it even results on partial loading (this is what happens for me with your model)

[image]

I faced the same issue for our humanoid robots

There is mostly two solutions I can propose you:
* Edit the collision shapes of small parts (you need OpenSCAD, see the README), and set `drawCollisions` to `true` in `config.json`, this way you can simplify shapes, or even 
* consider even ignoring some small parts from your design using the `ignore` section of `config.json` (see README.md)

Gregwar, 2019-07-26:
Also, as the warnings on the output are mentionning, you should assign materials to all your parts so that dynamics can be known for physics simulation

marshallmassengill, 2019-07-26:
Awesome!  We will work on adding in ignore flags and removing many of the
smaller parts.

Gregwar, 2019-07-26:
Ok
The "ignore" section will totally skip those parts, making the mass wrong but I don't have a better solution since
1. PyBullet only partially loads very big URDF files as I observed
2. PyBullet is really slow to load complex models anyway, so even if 1. would be fixed it would not be necessarly useable

marshallmassengill, 2019-07-26:
Yeah, that's kind of what we figured.  We can work on adjusting the mass
for what remains.  The main goal is to generate something that we can tweak
and expand hopefully.

Gregwar, 2019-07-26:
Doing approximation of shapes for collisions might help simulation as well

For example approximate an actuator with just a simple box is enough:
[image]

So the result looks like this if `drawCollisions` is `true`:
[image]

And like that if its `false`, and it is really slower:
[image]

Gregwar, 2019-11-20:
Hey, did you manage to import your robot?

marshallmassengill, 2019-11-20:
Yes!  Worked well enough for us.  Thanks for asking.

---Marshall

Gregwar, 2019-11-20:
Ok
Glad to hear that

closed 2019-11-20 by Gregwar