Rhoban/onshape-to-robot #12: Only part of dof is displayed in pybullet simulator issue

kimbring2, 2020-01-28:
Hello, thank you for making good tool.

I almost completed applying your program to the my own robot.
[image]

[image]

However, the problem that only two of the four wheels are displayed continued. Thus, I need some help for solving that problem.

[image]

Curiously, I confirmed that the first two of the revolulate mate, named dof_wheel, is displayed correctly. This is confirmed by a phenomenon that occurred in the same manner even when the order is changed.

Maybe I'd be grateful if you give me some advice for this phenomenon.

P.S. 
Here is my Onshape robot model link :  https://cad.onshape.com/documents/242e5d0f2f1cbff393c8e507/w/d559dc996c8b8bc22d12a621/e/61eb8c1b97972fad8186460d

Thank you,
From Dohyeong Kim

Gregwar, 2020-01-28:
Hello
My first guess would have been indeed the order of the mate connectors but it looks like they are similar for each joints

What about the output of the script run? Does it display that it found all the DOFs?
I can't execute it myself because the motors (PRT*) are not accessible to me via the API

Gregwar, 2020-01-28:
Update: i haven't seen your pyBullet screenshot, that suggest that the wheels are properly imported but just not showing (because there is indeed all DOFs)

This might be caused by pyBullet limitations for loading graphically big URDFs

Can you try to ignore some small parts and check if more parts are showing? 

I had similar issues on humanoid robot and I finally removed small parts and approximated everything using pure shapes (sphere, boxes, cylinders)

Gregwar, 2020-01-28:
Can you try as well with the version on the `urdf_fusion` branch?
https://github.com/Rhoban/onshape-to-robot/archive/urdf_fusion.zip

kimbring2, 2020-01-28:
Oh, thank you for kindness.

I can check all 4 dof is found and two not visible wheel existence in spite of there are transparent. 

Maybe I should decrease number of part because I also think there are so many part at motor and try urdf fusion branch.

I will tell you about result, 
From Kim

kimbring2, 2020-01-30:
Hello,

I check the reason of problem is number of part of my omniwheel. 
Becasue of complex structrue of omniwheel, I need to find a good way of reducing of number of part.

Thank you,
Dohyeong Kim

closed 2020-01-30 by kimbring2