Rhoban/onshape-to-robot #30: Support for MuJoCo (MJCF)

Gregwar, 2020-03-26:
Good point: MuJoCo seems to outperform other physics engine
Bad point: license is paid, so less reproductibility

But supporting MJCF (MuJoCo XML description format) would be feasible

@SammyRamone : do you use MuJoCo?

SammyRamone, 2020-03-26:
No, my team was mostly using Gazebo due to it's nice ROS integration, but it has some issues (mostly speed). I'm now starting to use PyBullet for reinforcement learning (many MuJoCo openai gym envs are also available for PyBullet). Until now I'm very happy with PyBullet and I also prefer open source, even if it is a bit slower. 
I'm also not sure if MuJoCo is faster because it is programmed better, or if they simplify the simulation to gain speed while spending accuracy.

renaudponcelet, 2024-01-19:
Any news regarding integrating MuJoCo ? At the moment I am doing onshape -> urdf -> MuJoCo. But for each step I lose information, and it needs a lot of modifications by hand. So I would be very interested in any information on this point :smile:

Gregwar, 2025-02-19:
Hello,
@renaudponcelet just pushed a new version supporting MuJoCo:
https://onshape-to-robot.readthedocs.io/en/latest/index.html

I plan to write more examples soon, hope it will make the process easier!

closed 2025-02-19 by Gregwar