Rhoban/onshape-to-robot #31: Any way to use without Onshape?

ugurkanates, 2020-04-17:
Since onShape is a very expensive and paid tool  not most of us have access to it. Can this extension to be generalised for CAD files without OnShape

Maybe something like Solidworks? I know there is an extension there to export for UDSF but that is not even similiar to what this repo does. ( Automatically finding tree structure etc)

marshallmassengill, 2020-04-17:
https://www.onshape.com/products/free

"At no cost, hobbyists, makers, and others who are willing to freely share their designs can pursue their projects without the hassles commonly found in old CAD systems. Onshape is accessible from any device, never loses data, and makes sharing and collaboration easy. For the ability to create private design data, see Onshape Professional."

So as long as your designs are publicly available then it's free.  They also have EDU plans: https://www.onshape.com/education-plan

ugurkanates, 2020-04-17:
Oh damn its super cool then. Sorry for inconvenience

closed 2020-04-17 by ugurkanates

Gregwar, 2020-04-17:
Indeed OnShape model is basically similar to GitHub, free for public projects, free for education & research and paid for the rest, fully online and so installation-less and cross-platform it makes it a great tradeoff for CAD

If you plan ahead to give a try to onshape-to-robot don't hesitate to give feedbacks on how it's going

ugurkanates, 2020-04-17:
I gave a try and while It retained shape of robot , textures were missing and it prints some empty output.

Though that is probably expected , it gave bunch of warnings about connecting some parts. As I said I have no knowledge about how these shapes work. As a PhD student I am trying to build a system that runs on pyBullet that I can manipulate via pyBullet function calls so I can do Deep Learning & Reinforcement Learning.

Sadly it seems it's not drag & drop. My json was like this , just copied readme.md one 

```
{
    "onshape_api": "https://cad.onshape.com",
    "onshape_access_key": "mynum",
    "onshape_secret_key": "mynum",
 
    "documentId": "d2f23f87102f359ea37c2198",
    "assemblyName": "IRB360_1_800_STD_3D_Flex Pikcer robot Assembly",
    "outputFormat": "urdf",
    "drawFrames": false,
    "drawCollisions": false,
    "useScads": true,
    "mergeSTLs" : true,
    "outputFormat":"urdf",
    "drawFrames": false,
    "noDynamics": false,
    "simplifySTLs": false,
    "maxSTLSize": 3,

    "jointMaxEffort": 1.5,
    "jointMaxVelocity": 22
}

```

My robot is IRB360 Flex Picker  which I found parts online via grabCAD ( although official site had parts there was no assembly version). 

https://cad.onshape.com/documents/d2f23f87102f359ea37c2198/w/f9912af4464908bd7f0e8425/e/8ccb14dc17224863c081d2e4

Sadly it seems not working directly. Although I found a version of robot that runs on  CoppeliaSim and has .ttm version .  It is kinda hard  for a software engineer  to learn all this mechanical terms relating to robot and modelling :) 

Still good repo for those who can manage it.
[image]
https://pasteboard.co/J4fYs5f.png

Gregwar, 2020-04-17:
If you want we can iterate until we get it running for you

Did you respect the assembly instructions in readme? Especially:
* naming dofs
* assigning materials

Does your robot has serial architecture? (Parallel joints are not handled by urdf and pyBullet anyway)

Feel free to send output to g.passault@gmail.com so I can have a look

Gregwar, 2020-04-17:
I can't open your OnShape assembly apparently, maybe permissions issues

Gregwar, 2020-04-17:
And it seems that your robot is parallel, there is currently no way to represent this using URDF since the representation is only a tree (no loops)

However there might be solutions to "serialize" it (make a serial representation that still has the same kinematic chain)

ugurkanates, 2020-04-17:
https://grabcad.com/library/irb360-1-800-flex-picker-parallel-robot-1

Upload for convenience to another file host/downloader no need to sign up
https://easyupload.io/m8kgzu
 
Its this one (just downloaded zip & uploaded into onShape without touching anything) 

While individual specific model isn't much important at the moment. My main objective at the moment to perfect grasp operation with this robot via deep RL. Any model could work it. Since I found this one I just went with it.( at grabCAD site, there is bunch of IR360 models)

Sure I will send you some details , I didnt did much about assembling process at onShape website. Since it pretty much said just upload ZIP and we will handle it.
(It just asks  a 3 choice if you want all in one file, seperate and a third option)

ugurkanates, 2020-04-17:
> And it seems that your robot is parallel, there is currently no way to represent this using URDF since the representation is only a tree (no loops)
> 
> However there might be solutions to "serialize" it (make a serial representation that still has the same kinematic chain)

Can you give me any source about how to achieve this. Maybe if you know a similiar repository or a blog post that talks detail about this. 

ugurkanates97@gmail.com btw.