Rhoban/onshape-to-robot #32: python modules missed?

AndreV84, 2020-04-17:
`./onshape-to-robot.py robots/myrobot
Traceback (most recent call last):
  File "./onshape-to-robot.py", line 4, in <module>
    from robot_description import RobotURDF, RobotSDF
  File "/home/nvidia/onshape-to-robot/robot_description.py", line 5, in <module>
    import stl_combine
  File "/home/nvidia/onshape-to-robot/stl_combine.py", line 5, in <module>
    import stl
  File "/home/nvidia/.local/lib/python2.7/site-packages/stl/__init__.py", line 2, in <module>
    import stl.ascii
ImportError: No module named ascii
`

` python3 -m onshape-to-robot.py robots/myrobot
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/nvidia/onshape-to-robot/onshape-to-robot.py", line 4, in <module>
    from robot_description import RobotURDF, RobotSDF
  File "/home/nvidia/onshape-to-robot/robot_description.py", line 5, in <module>
    import stl_combine
  File "/home/nvidia/onshape-to-robot/stl_combine.py", line 7, in <module>
    from stl import mesh
  File "/home/nvidia/.local/lib/python3.6/site-packages/stl/mesh.py", line 4, in <module>
    class Mesh(stl.BaseStl):
AttributeError: module 'stl' has no attribute 'BaseStl'`

Gregwar, 2020-04-17:
Did you install the following?

`pip install numpy-stl`

Also, maybe you have conflict with other library called stl?

AndreV84, 2020-04-17:
`pip uninstall stl`
changed the situation to the illustrated below:
python3 -m onshape-to-robot.py robots/myrobot
Traceback (most recent call last):
  File "/home/nvidia/.local/lib/python3.6/site-packages/commentjson/commentjson.py", line 168, in loads
    parsed = parser.parse(text)
  File "/home/nvidia/.local/lib/python3.6/site-packages/lark/lark.py", line 311, in parse
    return self.parser.parse(text, start=start)
  File "/home/nvidia/.local/lib/python3.6/site-packages/lark/parser_frontends.py", line 185, in parse
    return self._parse(text, start)
  File "/home/nvidia/.local/lib/python3.6/site-packages/lark/parser_frontends.py", line 54, in _parse
    return self.parser.parse(input, start, *args)
  File "/home/nvidia/.local/lib/python3.6/site-packages/lark/parsers/earley.py", line 292, in parse
    to_scan = self._parse(stream, columns, to_scan, start_symbol)
  File "/home/nvidia/.local/lib/python3.6/site-packages/lark/parsers/xearley.py", line 137, in _parse
    to_scan = scan(i, to_scan)
  File "/home/nvidia/.local/lib/python3.6/site-packages/lark/parsers/xearley.py", line 114, in scan
    raise UnexpectedCharacters(stream, i, text_line, text_column, {item.expect.name for item in to_scan}, set(to_scan))
lark.exceptions.UnexpectedCharacters: No terminal defined for '"' at line 10 col 5

    "jointMaxEffort": 1.5,
    ^

Expecting: {'RBRACE', 'COMMA'}

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/nvidia/.local/lib/python3.6/site-packages/commentjson/commentjson.py", line 203, in load
    return loads(fp.read(), **kwargs)
  File "/home/nvidia/.local/lib/python3.6/site-packages/commentjson/commentjson.py", line 171, in loads
    raise ValueError('Unable to parse text', text)
ValueError: ('Unable to parse text', '{\n    "documentId": "de3251d104c214dfe5847161",\n    "assemblyName": "robot",\n    "outputFormat": "urdf",\n    "drawFrames": false,\n    "drawCollisions": false,\n    "useScads": true,\n\n    "packageName": "robov2"\n    "jointMaxEffort": 1.5,\n    "jointMaxVelocity": 22\n}\n')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/nvidia/onshape-to-robot/onshape-to-robot.py", line 12, in <module>
    from load_robot import \
  File "/home/nvidia/onshape-to-robot/load_robot.py", line 3, in <module>
    from config import config, configFile
  File "/home/nvidia/onshape-to-robot/config.py", line 27, in <module>
    config = json.load(open(configFile))
  File "/home/nvidia/.local/lib/python3.6/site-packages/commentjson/commentjson.py", line 205, in load
    raise JSONLibraryException(e)
commentjson.commentjson.JSONLibraryException: JSON Library Exception

Exception thrown by library (json): ('Unable to parse text', '{\n    "documentId": "de3251d104c214dfe5847161",\n    "assemblyName": "robot",\n    "outputFormat": "urdf",\n    "drawFrames": false,\n    "drawCollisions": false,\n    "useScads": true,\n\n    "packageName": "robov2"\n    "jointMaxEffort": 1.5,\n    "jointMaxVelocity": 22\n}\n')

    Traceback (most recent call last):
      File "/home/nvidia/.local/lib/python3.6/site-packages/commentjson/commentjson.py", line 168, in loads
        parsed = parser.parse(text)
      File "/home/nvidia/.local/lib/python3.6/site-packages/lark/lark.py", line 311, in parse
        return self.parser.parse(text, start=start)
      File "/home/nvidia/.local/lib/python3.6/site-packages/lark/parser_frontends.py", line 185, in parse
        return self._parse(text, start)
      File "/home/nvidia/.local/lib/python3.6/site-packages/lark/parser_frontends.py", line 54, in _parse
        return self.parser.parse(input, start, *args)
      File "/home/nvidia/.local/lib/python3.6/site-packages/lark/parsers/earley.py", line 292, in parse
        to_scan = self._parse(stream, columns, to_scan, start_symbol)
      File "/home/nvidia/.local/lib/python3.6/site-packages/lark/parsers/xearley.py", line 137, in _parse
        to_scan = scan(i, to_scan)
      File "/home/nvidia/.local/lib/python3.6/site-packages/lark/parsers/xearley.py", line 114, in scan
        raise UnexpectedCharacters(stream, i, text_line, text_column, {item.expect.name for item in to_scan}, set(to_scan))
    lark.exceptions.UnexpectedCharacters: No terminal defined for '"' at line 10 col 5
    
        "jointMaxEffort": 1.5,
        ^
    
    Expecting: {'RBRACE', 'COMMA'}
    
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/home/nvidia/.local/lib/python3.6/site-packages/commentjson/commentjson.py", line 203, in load
        return loads(fp.read(), **kwargs)
      File "/home/nvidia/.local/lib/python3.6/site-packages/commentjson/commentjson.py", line 171, in loads
        raise ValueError('Unable to parse text', text)
    ValueError: ('Unable to parse text', '{\n    "documentId": "documentidherechanged",\n    "assemblyName": "robot",\n    "outputFormat": "urdf",\n    "drawFrames": false,\n    "drawCollisions": false,\n    "useScads": true,\n\n    "packageName": "robov2"\n    "jointMaxEffort": 1.5,\n    "jointMaxVelocity": 22\n}\n')

Gregwar, 2020-04-17:
Can you show your `config.json`?

Gregwar, 2020-04-17:
(Looks like you forgot a comma `,` the line just above)

AndreV84, 2020-04-17:
`{
    "documentId": "document-id-here",
    "assemblyName": "robot",
    "outputFormat": "urdf",
    "drawFrames": false,
    "drawCollisions": false,
    "useScads": true,

    "packageName": "name-here"
    "jointMaxEffort": 1.5,
    "jointMaxVelocity": 22
}
`

AndreV84, 2020-04-17:
added comma;
situation changed:
`python3 -m onshape-to-robot.py robots/myrobot
* Checking OpenSCAD presence...
Can't run openscad -v, disabling OpenSCAD support
TIP: consider installing openscad:
sudo add-apt-repository ppa:openscad/releases
sudo apt-get update
sudo apt-get install openscad

* Retrieving workspace ID ...
+ Using workspace id: alphanumeric-value-here

* Retrieving elements in the document, searching for the assembly...
ERROR: Unable to find assembly in this document
`

AndreV84, 2020-04-17:
what is the assembly that it can not find?

Gregwar, 2020-04-17:
The tool is searching for you robot assembly in your part studio file

It can be specified by `config.json` under `assemblyName` or you can remove this argument and the first assembly will be used

Gregwar, 2020-04-17:
(Since you can have as much parts or assemblies as you want in the same PartStudio document, you need to name the one that actually represents your robot)

AndreV84, 2020-04-17:
removed the assembly part of the config.json
`* Retrieving assembly
! ERROR (403) while using OnShape API
! {
  "moreInfoUrl" : "",
  "message" : "Ressource nicht vorhanden oder Sie haben keine Zugriffsberechtigung.",
  "status" : 403,
  "code" : 0
}
`

Gregwar, 2020-04-17:
Did you do that?
https://github.com/Rhoban/onshape-to-robot#setting-up-your-api-key

AndreV84, 2020-04-17:
` python3 -m onshape-to-robot.py robots/myrobot
* Checking OpenSCAD presence...
Can't run openscad -v, disabling OpenSCAD support
TIP: consider installing openscad:
sudo add-apt-repository ppa:openscad/releases
sudo apt-get update
sudo apt-get install openscad

* Retrieving workspace ID ...
+ Using workspace id: alphanumeric

* Retrieving elements in the document, searching for the assembly...
+ Found assembly, id: alphanumeric, name: "PRT_02_20_AL_1"
+ Found assembly, id: alphanumeric, name: "16T"
+ Found assembly, id: alphanumeric, name: "Assembly 1"

* Retrieving assembly
! ERROR (403) while using OnShape API
! {
  "moreInfoUrl" : "",
  "message" : "Ressource nicht vorhanden oder Sie haben keine Zugriffsberechtigung.",
  "status" : 403,
  "code" : 0
}
`

AndreV84, 2020-04-17:
I just sourced the bashrc
the situation changed:
`* Retrieving assembly
! ERROR (403) while using OnShape API
! {
  "moreInfoUrl" : "",
  "code" : 0,
  "message" : "Resource does not exist, or you do not have permission to access it.",
  "status" : 403
}
`

Gregwar, 2020-04-17:
Check out that this displays your keys:

```bash
echo $ONSHAPE_ACCESS_KEY
echo $ONSHAPE_SECRET_KEY
```

(of course, don't paste the result here)
If it doesn't work, check out this part of README:
https://github.com/Rhoban/onshape-to-robot#setting-up-your-api-key

AndreV84, 2020-04-17:
yes, they are in bashrc

Gregwar, 2020-04-17:
What assembly represents your robot that you try to export to URDF/SDF model?

Gregwar, 2020-04-17:
If it is `Assembly 1`, you should add thje following to your `config.json`:

```json
"assemblyName": "Assembly 1",
```

(Consider renaming it to something relevant for you)

AndreV84, 2020-04-17:
it is Assembly 1;
when I specified it it will throw permission issue;
do I have any other but viewer permission to do the conversion?
Thanks
`name: "Assembly 1"

* Retrieving assembly
! ERROR (403) while using OnShape API
! {
  "moreInfoUrl" : "",
  "message" : "Ressource nicht vorhanden oder Sie haben keine Zugriffsberechtigung.",
  "status" : 403,
  "code" : 0
}
`

AndreV84, 2020-04-17:
it first says that it found the assembly, but then will throw 403

Gregwar, 2020-04-17:
Yes, you mean you didn't ticked all the boxes in the dev portal?
I think you need more permissions than viewer indeed

Gregwar, 2020-04-17:
Assembly is found by API key but its contents can't be retrieved, means that you have some permissions but some others are missing I guess

AndreV84, 2020-04-17:
I have a viewer url from other guy;
it seems that I might need to clone the project or get other sort of elevation to continue

Gregwar, 2020-04-17:
I guess so, it would be interresting to investigate about wether it is possible to get enough informations with viewer accounts (since I guess everything is actually there in the document)

Gregwar, 2020-04-17:
Maybe it is because of the workspace, can you try setting the `versionId` ?

AndreV84, 2020-04-17:
it got processing with elevating of access level;
thank you for guidelines

Gregwar, 2020-04-17:
Good, hope everything will work for you

AndreV84, 2020-04-17:
it worked for a while then has thrown:
`
* Writing URDF file
/usr/bin/python3: Error while finding module specification for 'onshape-to-robot.py' (AttributeError: module 'onshape-to-robot' has no attribute '__path__')
`

AndreV84, 2020-04-17:
without -m flag  it has written the output file

AndreV84, 2020-04-17:
`./bullet.py robots/myrobot
pybullet build time: Apr 16 2020 20:52:31
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=NVIDIA Tegra Xavier (nvgpu)/integrated
GL_VERSION=3.3.0 NVIDIA 32.3.1
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 32.3.1
Vendor = NVIDIA Corporation
Renderer = NVIDIA Tegra Xavier (nvgpu)/integrated
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = NVIDIA Corporation
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/UrdfFindMeshFile.h,112]:

b3Printf: robots/myrobot/robot.urdf:6: cannot find 'robov2dummy_dc13_idle.stl' in any directory in urdf path

Warning: b3Error[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,121]:

Warning: Could not parse visual element for Link:
Warning: b3Error[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,121]:

Warning: 51mmslew_1
Warning: b3Error[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,121]:

Warning: failed to parse link
Traceback (most recent call last):
  File "./bullet.py", line 11, in <module>
    sim = Simulation(directory)
  File "/home/nvidia/onshape-to-robot/simulation.py", line 24, in __init__
    cubeStartPos, cubeStartOrientation)
pybullet.error: Cannot load URDF file.
ven = NVIDIA Corporation
numActiveThreads = 0
stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
`

AndreV84, 2020-04-17:
does it error because openscad is not installed?

AndreV84, 2020-04-17:
should it work if I use "outputFormat": "sdf" in config.json?
`Traceback (most recent call last):
  File "onshape-to-robot.py", line 185, in <module>
    buildRobot(tree, np.matrix(np.identity(4)))
  File "onshape-to-robot.py", line 143, in buildRobot
    addPart(occurrence, matrix)
  File "onshape-to-robot.py", line 99, in addPart
    robot.addPart(pose, config['outputDirectory']+'/'+stlFile, mass, com, inertia, color, shapes, prefix)
  File "/home/nvidia/onshape-to-robot/robot_description.py", line 376, in addPart
    if linkName is not None:
NameError: name 'linkName' is not defined
`

Gregwar, 2020-04-17:
This issue is a bug (I'm looking)

However I don't understand why your robot doesn't load in pyBullet, can you share a zip of the output somehow?

Gregwar, 2020-04-17:
Looks like the `stl` file `robov2dummy_dc13_idle.stl` can't be found
OpenScad is not required if you don't want to do pure shape approximation

Gregwar, 2020-04-17:
Are STL files outputed in your target directory?

AndreV84, 2020-04-17:
it seems that the package name hasn't got appended to the beginning of the fines;
either I can ```cp dummy_dc13_idle.stl robov2dummy_dc13_idle.stl 
```
or rebuild them after deleting package name from config.json
`/onshape-to-robot/robots/myrobot$ ls
51mmslew.stl               arm2_horizontal_plate.stl  arm4.stl        base.stl             part_12.stl  part_5.stl
615238.stl                 arm2.stl                   base_gear.stl   bottom_hand.stl      part_1.stl   part_6.stl
615262.stl                 arm3_horizontal_plate.stl  basejoint.stl   config.json          part_26.stl  part_8.stl
adapter.stl                arm3.stl                   base_plate.stl  dummy_dc13_idle.stl  part_2.stl   robot.urdf
arm2_conencting_plate.stl  arm4_horizontal_palte.stl  base_side.stl   first_arm.stl        part_32.stl  zaxis_gear.stl`

Gregwar, 2020-04-17:
Do you mean the stl name in your urdf is not consistent with the one on the disk?
Can you paste the `<visual>` tag from your URDF?

AndreV84, 2020-04-17:
`<visual>
<origin xyz="0.0231264 0.1161 0.346124" rpy="1.5708 1.5708 0" />
<geometry>
<mesh filename="package://adapter.stl"/>
</geometry>
<material name="adapter_material">
<color rgba="0.615686 0.811765 0.929412 1.0"/>
</material>
</visual>
`
there are many these tags

Gregwar, 2020-04-17:
There is one reffering to `robov2dummy_dc13_idle.stl` ?

Gregwar, 2020-04-17:
Did you defined a packageName? This is related to SDFs only, if exporting URDF remove it from your JSON

AndreV84, 2020-04-17:
there is first atempt with package name 
https://drive.google.com/file/d/17QHbW27pAZSvhWDUQuaFlH_QvjKllRYo/view?usp=sharing
there is second attempt without it
https://drive.google.com/file/d/1yHaslvO2Oc23JnlTvlbRw3EPKvRGMNww/view?usp=sharing

AndreV84, 2020-04-17:
if you request sharing through the URL I can share the resulting zip file

AndreV84, 2020-04-17:
bullet py will run, but empty

Gregwar, 2020-04-17:
> if you request sharing through the URL I can share the resulting zip file

Ok I requested access

AndreV84, 2020-04-17:
export to sdf won't work
shared

Gregwar, 2020-04-17:
Actually it is there, you can see it in wireframe (pressing `w` in pyBullet)
But visuals are not showing up, I'm checking why
There is no joints as well (so it's like one big rigid body)

AndreV84, 2020-04-17:
très bien merci beaucoup

Gregwar, 2020-04-17:
Ok looks like some pyBullet related issue (too much visual tags I guess)
Can you install meshlab and add:

```json
"mergeSTLs": true
```

In your json?

Gregwar, 2020-04-17:
Also, there is no DOFs in this robot (check the README file for information on how you can add DOFs in your assembly)

Gregwar, 2020-04-17:
Add both:

```json
"simplifySTLs": true,
"mergeSTLs": true
```

This will result in only one visual link containing one combined more lightweight STL

AndreV84, 2020-04-17:
installed meshlab, added both
mergeSTL
simplifySTL
` Adding part arm4 <1>
Traceback (most recent call last):
  File "onshape-to-robot.py", line 185, in <module>
    buildRobot(tree, np.matrix(np.identity(4)))
  File "onshape-to-robot.py", line 144, in buildRobot
    robot.endLink()
  File "/home/nvidia/onshape-to-robot/robot_description.py", line 190, in endLink
    stl_combine.save_mesh(self._mesh, self._mesh_dir+'/'+self._link_name+'.stl')
  File "/home/nvidia/onshape-to-robot/stl_combine.py", line 14, in save_mesh
    mesh.save(stl_file, mode=stl.Mode.BINARY)
AttributeError: module 'stl' has no attribute 'Mode'
`

Gregwar, 2020-04-17:
Again issue with `stl` library?

```python
imporPython 3.6.7 (default, Oct 22 2018, 11:32:17) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.5.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import stl                                                                    

In [2]: stl.Mode.BINARY                                                               
Out[2]: <Mode.BINARY: 2>
```

Gregwar, 2020-04-17:
You can consider making a virtualenv with only the dependencies:

 ```
pip install numpy pybullet requests commentjson colorama numpy-stl
```

AndreV84, 2020-04-17:
for DOF this readme file?
https://github.com/Rhoban/onshape-to-robot/blob/master/README.md
```python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import stl
```
creating virtualenv

AndreV84, 2020-04-17:
` python3
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import stl
>>> stl.Mode.BINARY
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'stl' has no attribute 'Mode'
>>> 
`

Gregwar, 2020-04-17:
Yes this:
https://github.com/Rhoban/onshape-to-robot/blob/master/README.md#design-time-considerations

Did you do:
```
pip install numpy-stl
```

?

AndreV84, 2020-04-17:
previously I did but then I uninstalled it;
in venv I won't install it

AndreV84, 2020-04-17:
in virtualenv it builds urdf, doesn't build sdf;
I shall open the original source and rename of to match the format dof_something;

Gregwar, 2020-04-17:
What is your issue with SDF? If it's because of `linkName`  error, try to `git pull`
For the DOFs name, yes you have to change the naming and observe some convention to have this working (mate order matters as well, check out the README)

AndreV84, 2020-04-17:
SDF builds

AndreV84, 2020-04-17:
` gazebo
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
gzclient: /build/ogre-1.9-i02lBV/ogre-1.9-1.9.0+dfsg1/OgreMain/include/OgreAxisAlignedBox.h:252: void Ogre::AxisAlignedBox::setExtents(const Ogre::Vector3&, const Ogre::Vector3&): Assertion `(min.x <= max.x && min.y <= max.y && min.z <= max.z) && "The minimum corner of the box must be less than or equal to maximum corner"' failed.
escalating to SIGKILL on server
`
gazebo fails when clicking on imported model

[9 more replies]

closed 2020-10-14 by Gregwar