Rhoban/onshape-to-robot #33: PackageName in collisions STL?

Gregwar, 2020-04-17:
Ping @SammyRamone , I don't know the exact purpose of packageName (i expect it is about defining the path of your STLs?), but isn't it a bug that it is not also appended in the STLs files that are uses for collisions (assuming you don't use pure shapes) ?

SammyRamone, 2020-04-17:
You expected correctly. The packageName is automatically resolved by ROS to the path of the ROS package with this name. The standard is to have a package with the URDF of your robot named ROBOTNAME_description (e.g. in our case wolfgang_description). In this package holds all STL files and the URDF. This way you never have to care about where the files are actually in your file system, as ROS takes care of it.
In my URDF (https://github.com/bit-bots/wolfgang_robot/blob/master/wolfgang_description/urdf/robot.urdf) it works correctly for visuals and collisions.
It is added here for both
https://github.com/Rhoban/onshape-to-robot/blob/661093226ed8f4a0af3edb0fa25bdb83c85925a4/robot_description.py#L242

If you are talking about SDF, I don't know. I don't use them currently and I'm also not sure how ROS handles them.

Gregwar, 2020-04-17:
Ok so packageName is mostly a path ending with `/` I guess
thanks for information

closed 2020-04-17 by Gregwar