Rhoban/onshape-to-robot #35: Using URDF Inertia in Simulation

SammyRamone, 2020-05-14:
Just a small thing. When loading the URDF into bullet it does not use the provided Inertias but recomputes them:

> by default, Bullet recomputed the inertia tensor based on mass and volume of the collision shape. If you can provide more accurate inertia tensor, use this flag.

 You need to set the URDF_USE_INERTIA_FROM_FILE flag to prevent this here https://github.com/Rhoban/onshape-to-robot/blob/baef494317cdede80ad2a00b016801fd08b6d733/onshape_to_robot/simulation.py#L70-L72

I think it would be good to change this in the simulation, as people may get confused why the model is behaving incorrectly after adding good inertia values in the config using the "dynamics" parameter.

Gregwar, 2020-05-14:
Ok, I agree this should be done
However having a look at the pybullet code itself I am not quite sure of the actual effect of this flag
I mean, if the collisions are pure meshes, are inertia really re-computed from scratch by pyBullet?

Still you are right I should add it

Gregwar, 2020-05-14:
So far, `simulation.py` is not documented as well, and I think it should be

This is what I am actually doing now
```python
from onshape_to_robot.simulation import Simulation
```

And using this in gyms as a wrapper around pybullet to make experiments

SammyRamone, 2020-05-14:
Yeah I was also wondered why PyBullet tries to compute any inertia if its already in the URDF. I just saw this in the documentation (https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/preview#heading=h.sbnykoneq1me) and did not further look into the code.
I'm using my own simulation.py (because I programmed it already some time ago). But some documentation for the simulation would be nice (the new documentation structure is nice btw) to give others an easier start.