Rhoban/onshape-to-robot #39: joint damping and friction

SammyRamone, 2020-06-10:
I implemented additional config values to specify joint damping and friction for URDFs on my fork (see https://github.com/bit-bots/onshape-to-robot/commit/ef9db6e9ff0e816150fb932ce8089c27f8b77d38). The implementation is basically the same as for the joint velocity and torque.
Are you interested in having this upstream? If yes, I can do a PR or you can just copy the few lines of code.

lokiledev, 2021-04-13:
Upvoting this it's a nice feature!

Niklas-Mulsow, 2023-09-01:
Upvoting this it's a nice feature!

Gregwar, 2023-09-01:
This is related to #108, all joints-related configuration could be uniformized, including friction and damping would then be smoother

Gregwar, 2025-02-19:
This is now possible using `joint_properties` for URDF:
https://onshape-to-robot.readthedocs.io/en/latest/exporter_urdf.html

And MuJoCo:
https://onshape-to-robot.readthedocs.io/en/latest/exporter_mujoco.html

closed 2025-02-19 by Gregwar