Rhoban/onshape-to-robot #46: Error 404 while using OnShape API

neoalejo, 2021-02-17:
Hello

I am trying to create a urdf file from an onshape model of a robot manipulator arm, and I am getting the following error:

* Checking OpenSCAD presence...
* Checking MeshLab presence...

* Retrieving workspace ID ...
+ Using workspace id: dfe513c099f86b8f8b4b33c1

* Retrieving elements in the document, searching for the assembly...
+ Found assembly, id: e7806a8b6e8ca46848b0ad73, name: "Bravo_7"

* Retrieving assembly "Bravo_7" with id e7806a8b6e8ca46848b0ad73

* Getting assembly features, scanning for DOFs...
+ Found DOF: j7 (revolute)[0.087: 6.196]
+ Found DOF: j6 (revolute)[0.0: 3.142]
+ Found DOF: j5 (revolute)[0.0: 3.142]
+ Found DOF: j4 (revolute)[0.087: 6.196]
+ Found DOF: j3 (revolute)[0.0: 3.142]
WARNING: joint dof_j2 of type revolute has no limits 
+ Found DOF: j2 (revolute)
* Found total 6 DOFs

* Building robot tree
* Trunk is Base <1>
* Adding top-level instance [Base <1>]
+ Adding part RS2-276-def <1>
+ Adding part RS2-266-def <1>
* Adding top-level instance [Link1 <1>]
+ Adding part RS2-249-def <1>
+ Adding part RS2-217-def <1>
* Adding top-level instance [Link2 <1>]
+ Adding part RS2-236-def <1>
+ Adding part RS2-214-def <1>
+ Adding part RS2-212-def <1>
+ Adding part RS2-250-def <1>
! ERROR (404) while using OnShape API
! <!doctype html>
<html lang="en" ng-app="Woolsthorpe" ng-csp>
  <head>
    <meta charset="utf-8">
    <meta name="robots" content="noindex,nofollow" />
    
    <meta name="apple-itunes-app" content="app-id=923421284"/>
    <meta name="description" content="Login / Signin to Onshape - the modern CAD system used by thousands of companies worldwide."/>
    <link rel="shortcut icon" href="/favicon.png">
    <title ng-bind="$root.title">Onshape</title>

    <link rel="stylesheet" type="text/css" href="/css/jquery-ui/jquery-ui-1.10.4.custom.css" />
    <link rel="stylesheet" type="text/css" href="/css/jquery.contextMenu.css" />
    <link rel="stylesheet" type="text/css" href="/css/glyphicons.css" />
    <link rel="stylesheet" type="text/css" href="/css/demoserver-bootstrap.css" />
    <link rel="stylesheet" type="text/css" href="/css/jquery.nouislider.css" />
    <link rel="stylesheet" type="text/css" href="/css/colorpicker-themes.css" />

    <link rel="stylesheet" href="/css/bootstrap/bootstrap.css"/>
    <link rel='stylesheet' type='text/css' href='/css/fonts.css' />

      <link href="/css/newton.css" rel="stylesheet">

      <script src="/js/runtime.31d2b0b55f004549b961.js"></script>
      <script src="/js/vendor-bundle.31d2b0b55f004549b961.js"></script>
      <script src="/js/webpack-vendor.31d2b0b55f004549b961.js"></script>
      <script src="/js/woolsthorpe.31d2b0b55f004549b961.js"></script>
      <script src="/css/newton.31d2b0b55f004549b961.js"></script>

    <script type="text/javascript" src="https://js.stripe.com/v2/" async></script>
    <script type="text/javascript" src="https://www.zuora.com/apps/Hosted/lite2/js/HPM2Security.js" async></script>
  </head>
<body os-application-fullframe>

<div ng-include="'/images/icons.min.svg'" class="os-svgmin-container"></div>
<div ui-view class="os-app-container"></div>
<div ng-include="'/project/web/ts/user-notifications/user-notifications.html'"></div>
<div os-usersnap></div>
<div os-zendesk></div>
<os-message-bubble></os-message-bubble>
</body>
</html>

I tried to isolate the error and I believe it has something to do with Link2 of my model, but I am not sure what can the error with this part of the robot be. Can somebody explain me what does the html part of the error means? because a lot of the text does not make any sense.

Here is the link to my onshape document:

https://cad.onshape.com/documents/6fdce4c712a1f625a0e4df69/w/dfe513c099f86b8f8b4b33c1/e/5bd485de8a70d7474f8490c3

This is the isolated link of the arm that I believe is causing the problem:

https://cad.onshape.com/documents/c7f4343ab134d605156724b1/w/35a6d286ff1923733e540600/e/17437e1268bcb23d70d88912

Thank you in advance

Gregwar, 2021-02-17:
Hello,

I believe this issue happens because your assembly is referencing parts and you don't have edition access to it

In this case, the following URL is invoked:
https://cad.onshape.com/api/parts/d/c7f4343ab134d605156724b1/m/333ac268f2d757288de7bccb/e/b47021706b0771d8cc85e3a7/partid/JF//metadata?configuration=default

Which triggers an access denied.

I am not sure why there are privileges required since this should be a read-only API request, I will open an issue in OnShape tickets system itself to investigate. In the meantime, a solution for you would be to make a copy of this document and use it instead.

Gregwar, 2021-02-17:
I opened issue on OnShape bug tracking system and I will give updates here

It is weird that this works:
https://cad.onshape.com/api/parts/d/c7f4343ab134d605156724b1/m/333ac268f2d757288de7bccb/e/b47021706b0771d8cc85e3a7/

And this doesn't:
https://cad.onshape.com/api/parts/d/c7f4343ab134d605156724b1/m/333ac268f2d757288de7bccb/e/b47021706b0771d8cc85e3a7/partid/JF//metadata?configuration=default

Since they are both GET requests doing only read operations

Gregwar, 2021-02-17:
Hello,

The problem was that onshape internal partids can contain slash (here `JF/ `)
I fixed the code to escape those strange slashes, if you update to 0.3.6 it should work

neoalejo, 2021-02-18:
Thank you very much! the fix solved the problem! :D

closed 2021-02-18 by neoalejo

agillies8, 2022-03-18:
I am now seeing a similar issue with v 0.3.16:

`
https://cad.onshape.com/api/parts/d/......../partid/JF%2f/massproperties?configuration=default
! ERROR (404) while using OnShape API
! {
  "moreInfoUrl" : "",
  "message" : "Not found.",
  "status" : 404,
  "code" : 0
}
`
When I check the partid directly via the api, it appears as "JF/"

Could this bug be lingering somewhere still?

Gregwar, 2022-03-18:
Hello,
Maybe, that would be simplier if I had an example that I could reproduce myself, is it a project you can share with me?

noelhwang, 2022-11-04:
Unfortunately, I can't share my project, but I wanted to mention I also had a 404 error show up - when clicking through the URL generated by the error message, the partid portion of the url was malformed via the text implemented in this hotfix. i.e. "/" replaced by "%252f". To test, I manually replacing the malformed text in the partid with "/", and the link went through just fine. My workaround for the time being was to run from source on the commit just prior to the change.

MrKeith99, 2022-11-15:
ive experienced the same problem

Gregwar, 2022-11-15:
@MrKeith99 I can see your design but I can't import it because I am not allowed to copy it for instance
Maybe you can isolate the part that causes the problem and share that one specifically ?

I will try to create a document with a lot of parts to see if I can get one generated with a slash to add it to the tests/examples

Gregwar, 2022-11-15:
I could reproduce by adding 64 parts to a document and running the import to have a lot of partids
Looks like escaping the / only once in the metadata request does the trick.

I re-packaged `0.3.17`, can you upgrade and tell me if it works now?

MrKeith99, 2022-11-15:
```
pybullet build time: May 20 2022 19:44:17
* Checking OpenSCAD presence...
* Checking MeshLab presence...

* Retrieving workspace ID ...
+ Using workspace id: 88fc4f32ca2426ef06393394

* Retrieving elements in the document, searching for the assembly...
+ Found assembly, id: 294230932b42f016a12a1535, name: "SOBIT_PRO(MK-4)"
+ Found assembly, id: c6735a901ea2e1529c215af9, name: "arm_elbow_lower_pan"
+ Found assembly, id: 61c5dbf2e78867df6664c5de, name: "arm_elbow_lower_tilt"
+ Found assembly, id: 3a0f7d17aa2f274df5985e96, name: "platform_bottom"
+ Found assembly, id: bdb70e69e307e5054af63303, name: "platform_middle"
+ Found assembly, id: 3e15eb4f7e265186450c15c6, name: "platform_top"
+ Found assembly, id: c1ee2b20f8570e74b87b79f5, name: "urg"
+ Found assembly, id: a29412368500b01c0df2c4b3, name: "pole"
+ Found assembly, id: dbe54c2c550b203528d8f05e, name: "mic"
+ Found assembly, id: baecf1fcd422834977eda067, name: "emergency button"
+ Found assembly, id: 8809880f55ab8e040f597c43, name: "platform"
+ Found assembly, id: 8fa164ab5d61172304b415f1, name: "head_camera"
+ Found assembly, id: 431229acf4c8f72bf9609f27, name: "arm"
+ Found assembly, id: 1d05f411dc80d6cac0d8e13b, name: "wheel"

* Retrieving assembly "wheel" with id 1d05f411dc80d6cac0d8e13b

* Getting assembly features, scanning for DOFs...
* Found total 0 DOFs

* Building robot tree
* Trunk is SOBIT_wheel v1 <1>
* Adding top-level instance [SOBIT_wheel v1 <1>]
+ Adding part X-430_IDLE <1>
WARNING: part X-430_IDLE <1> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <7>
WARNING: part X-430_SPACER_RING <7> has no mass, maybe you should assign a material to it ?
+ Adding part SOBIT_wheel v1 <8>
WARNING: part SOBIT_wheel v1 <8> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_CVR_CABLE <1>
WARNING: part X-430_CVR_CABLE <1> has no mass, maybe you should assign a material to it ?
+ Adding part SOBIT_wheel v1 <1>
WARNING: part SOBIT_wheel v1 <1> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <6>
WARNING: part X-430_SPACER_RING <6> has no mass, maybe you should assign a material to it ?
+ Adding part SOBIT_wheel v1 <3>
WARNING: part SOBIT_wheel v1 <3> has no mass, maybe you should assign a material to it ?
+ Adding part SOBIT_wheel v1 <4>
WARNING: part SOBIT_wheel v1 <4> has no mass, maybe you should assign a material to it ?
+ Adding part SOBIT_wheel v1 <5>
WARNING: part SOBIT_wheel v1 <5> has no mass, maybe you should assign a material to it ?
+ Adding part SOBIT_wheel v1 <9>
WARNING: part SOBIT_wheel v1 <9> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <4>
WARNING: part X-430_SPACER_RING <4> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <4>
WARNING: part X-430_SPACER_RING <4> has no mass, maybe you should assign a material to it ?
+ Adding part SOBIT_wheel v1 <7>
WARNING: part SOBIT_wheel v1 <7> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <5>
WARNING: part X-430_SPACER_RING <5> has no mass, maybe you should assign a material to it ?
+ Adding part SOBIT_wheel v1 <6>
WARNING: part SOBIT_wheel v1 <6> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <3>
WARNING: part X-430_SPACER_RING <3> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <3>
WARNING: part X-430_SPACER_RING <3> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <7>
WARNING: part X-430_SPACER_RING <7> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_IDLE <1>
WARNING: part X-430_IDLE <1> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <5>
WARNING: part X-430_SPACER_RING <5> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <6>
WARNING: part X-430_SPACER_RING <6> has no mass, maybe you should assign a material to it ?
+ Adding part SOBIT_wheel v1 <2>
WARNING: part SOBIT_wheel v1 <2> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <2>
WARNING: part X-430_SPACER_RING <2> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <2>
WARNING: part X-430_SPACER_RING <2> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <8>
WARNING: part X-430_SPACER_RING <8> has no mass, maybe you should assign a material to it ?
+ Adding part fr12_h101 v1 <1>
WARNING: part fr12_h101 v1 <1> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <8>
WARNING: part X-430_SPACER_RING <8> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <1>
WARNING: part X-430_SPACER_RING <1> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_CVR_CABLE <1>
WARNING: part X-430_CVR_CABLE <1> has no mass, maybe you should assign a material to it ?
+ Adding part X-430_SPACER_RING <1>
WARNING: part X-430_SPACER_RING <1> has no mass, maybe you should assign a material to it ?

* Writing URDF file

```

I have been able to pass the path problem but unfortunately it seems to stop in the "wheel" assembly, so the other parts are not retrieved

MrKeith99, 2022-11-15:
btw, this is the correct [link](https://cad.onshape.com/documents/af4023761ee3c75dbf467d0d/w/88fc4f32ca2426ef06393394/e/294230932b42f016a12a1535?renderMode=0&uiState=637360972cd11d64bcc3227e) to the project

Gregwar, 2022-11-15:
You can specify which assembly you want to export using `assemblyName` in the `config.json` file, see:
https://onshape-to-robot.readthedocs.io/en/latest/config.html#assemblyname

RyotaYoneyama, 2023-02-04:
Hello, I also encountered with same problem saying
`! ERROR (404) while using OnShape API
! {
  "message" : "Not found.",
  "moreInfoUrl" : "",
  "status" : 404,
  "code" : 0
}
`

My cad link is :
[https://cad.onshape.com/documents/3f8fd3cb66199d54e5d8a8ef/w/dd6d2618a85894306c9569e8/e/b3f3a1e5acf7b491d180e6c6](link)  

And the version of onshape-to-robot 0.3.17.

I had not have a problem last week but now I got it.

Gregwar, 2023-02-04:
Looks like API for metadata changed. 
I switched to using `/metadata/...` instead, and released `0.3.18`, @RyotaYoneyama can you give it a try?

Gregwar, 2023-02-04:
[image]

Works for me

RyotaYoneyama, 2023-02-05:
Hi, thank you for your response @Gregwar .
It worked almost perfectly!
However, I got an error while using mergeSTLs and simplifySTLs simultaneously although it is not big deal for me.
The error message is 

`+ joint_0_visual.stl is 6.56 M, running mesh simplification
Traceback (most recent call last):
  File "/home/ryota/.local/bin/onshape-to-robot", line 8, in <module>
    sys.exit(onshape_to_robot.main())
  File "/home/ryota/.local/lib/python3.10/site-packages/onshape_to_robot/onshape_to_robot.py", line 242, in main
    buildRobot(tree, np.matrix(np.identity(4)))
  File "/home/ryota/.local/lib/python3.10/site-packages/onshape_to_robot/onshape_to_robot.py", line 207, in buildRobot
    robot.endLink()
  File "/home/ryota/.local/lib/python3.10/site-packages/onshape_to_robot/robot_description.py", line 221, in endLink
    stl_combine.simplify_stl(self.meshDir+'/'+filename, self.maxSTLSize)
  File "/home/ryota/.local/lib/python3.10/site-packages/onshape_to_robot/stl_combine.py", line 93, in simplify_stl
    reduce_faces('/tmp/simplify.stl', stl_file, max_size / size_M)
  File "/home/ryota/.local/lib/python3.10/site-packages/onshape_to_robot/stl_combine.py", line 80, in reduce_faces
    output = subprocess.check_output(command, shell=True)
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'meshlabserver -i /tmp/simplify.stl -s /tmp/filter_file_tmp.mlx -o wormy_arm_description/urdf/raw/joint_0_visual.stl -om vn fn > /tmp/meshlab.log 2>&1' returned non-zero exit status 255.
onshape-to-robot wormy_arm_description/urdf/raw  7.55s user 5.75s system 302% cpu 4.401 total
`