Rhoban/onshape-to-robot #47: Pulling from Branches

ThomasPiechulek, 2021-02-25:
Hello,
i can not find a possibility to onshape-to-robot on a specified branch. The script seems to just take the last one active.
Can you add the Workspace Id in the Config.json to solve this ?

Gregwar, 2021-02-25:
Hello,
Can you use `versionId` parameter for that ?
https://onshape-to-robot.readthedocs.io/en/latest/config.html#versionid

Gregwar, 2021-02-25:
Currently, `versionId` can be used to specify a version (`/v/`), I can also add a `workspaceId` to be used `/w/` if you need it

ThomasPiechulek, 2021-02-26:
Hello,
yes that would be great.
Beacause we try to do everything automatically, and with the version id, we have to manually create versions every time...
Thank you so much.

Gregwar, 2021-02-26:
Just added `workspaceId` configuration entry, can you confirm it's working for you ?

ThomasPiechulek, 2021-03-01:
works fine, thank you.

ThomasPiechulek, 2021-03-02:
In the code we are searching for the ID, but it wants to print the Keyword Workspace, so some kollegues have this error occuring.
Strangely this Eroor does not appear for me so sorry for the early confirm.

Traceback (most recent call last):
  File ".../bin/onshape-to-robot", line 3, in <module>
    from onshape_to_robot import onshape_to_robot
  File "...lib/python3.8/site-packages/onshape_to_robot/onshape_to_robot.py", line 14, in <module>
    from .load_robot import \
  File ".../lib/python3.8/site-packages/onshape_to_robot/load_robot.py", line 19, in <module>
    config['workspace']+' ...' + Style.RESET_ALL)
KeyError: 'workspace'

isetre, 2021-03-02:
The problem seems to be in the load_robot.py file at ll. 18 ff., where the config is accessed with a faulty key and later on the workspaceId variable is required. The code snippet below contains my edits with which the code worked.
`workspaceId=config['workspaceId']
print("\n" + Style.BRIGHT + '* Using configuration workspace ID ' +
          config['workspaceId']+' ...' + Style.RESET_ALL)`
By the way, great work you're doing here!

Gregwar, 2021-03-02:
You are right, sorry
Should be fixed in `v0.3.9`

isetre, 2021-03-02:
works fine now, thanks

closed 2025-02-20 by Gregwar