Rhoban/onshape-to-robot #48: WARNING: part X has no dynamics (maybe it is a surface)

kschyschtov, 2021-03-10:
Hello,
Last year in December I generated correctly the urdf from my onshape model which contained faulty topology (although it looks good visually). The newest version of onshape-to-robot that I tried a few days ago shows a problem: WARNING: part X has no dynamics (maybe it is a surface). It creates the urdf file but without linking to an stl model. I generated again a correct urdf with v.0.3.4 so definitely something changed between the newest and that version. I am not sure if this behaviour was desired but at least it prevents me from using the newest version because I cannot change the imported model in onshape and I cannot find what causes the faulty topology.

kschyschtov, 2021-03-10:
Thanks for the prompt response. Not to share the link publicly, I sent it to your email.

The correctly generated link in urdf looks like this: 
<link name="blmarm_-_j1">
<visual>
<origin xyz="1.28786e-14 0 0" rpy="8.40762e-16 -1.22125e-15 8.63267e-40" />
<geometry>
<mesh filename="package://blmarm_-_j1.stl"/>
</geometry>
<material name="blmarm_-_j1_material">
<color rgba="0.768627 0.886275 0.952941 1.0"/>
</material>
</visual>
<collision>
<origin xyz="1.28786e-14 0 0" rpy="8.40762e-16 -1.22125e-15 8.63267e-40" />
<geometry>
<mesh filename="package://blmarm_-_j1.stl"/>
</geometry>
<material name="blmarm_-_j1_material">
<color rgba="0.768627 0.886275 0.952941 1.0"/>
</material>
</collision>
<inertial>
<origin xyz="-0.0220872 -0.00289466 -0.0953669" rpy="0 0 0"/>
<mass value="27.0338" />
<inertia ixx="0.0980756" ixy="-0.00451259"  ixz="-0.0107472" iyy="0.15549" iyz="-0.00116727" izz="0.144397" />
</inertial>
</link>

while the newest version generates this:
<link name="blmarm_-_wagon">
<inertial>
<origin xyz="0 0 0" rpy="0 0 0"/>
<mass value="0" />
<inertia ixx="0" ixy="0"  ixz="0" iyy="0" iyz="0" izz="0" />
</inertial>
</link>

Gregwar, 2021-03-10:
I did a fix so that `v3.0.10` should work for you, but I still need to understand why

(The `partId` can contain slashes `/` which need to be escaped for some API words and not others)

neoalejo, 2021-03-11:
If it helps, I also had this issue just after trying the version that solved issue #46

kschyschtov, 2021-03-11:
Thanks for looking into it.

I upgraded the onshape-to-robot with command 
`python3 -m pip install --upgrade onshape-to-robot`
and it installed onshape-to-robot-0.3.10.

But unfortunately, there is still the same problem with the part without the link to STL file nor other properties related to shape.

Gregwar, 2021-03-16:
Hello

Indeed the STL API word also requires no escaping for slashes
Can you try `v0.3.11` ? :-)

kschyschtov, 2021-03-16:
Still the same ;)

Gregwar, 2021-03-16:
With current build I can load this in pyBullet, even if it is highly unstable:

[image]

This isn't working for you ?

kschyschtov, 2021-03-16:
It still generates the wagon link with zeros. I uninstalled and installed again the newest version, the same. Should I do anything else to update?

Gregwar, 2021-03-17:
Try to clear your cache with `onshape-to-robot-clear-cache` and delete STL files if they exist and are empty

kschyschtov, 2021-03-18:
Yes! Clearing cache helped. It seems to generate correctly now, thanks!

Gregwar, 2021-03-19:
Ok, great!
I'll close this, don't hesitate to re-open or create another issue later

closed 2021-03-19 by Gregwar