Rhoban/onshape-to-robot #49: Origin changes on Assembly copy

ThomasPiechulek, 2021-03-12:
Dear Onshape-Team,

 

I had a problem with your product.

 

In our project, we want to build a constrained model for machine learning.
To solve this, I copied the main assembly and adjusted the joint limits.

 

When exporting the assembly with onshape-to-robot, we experienced changes in the origins of the joint tag.
This lead to an erroneous behaviour in our simulation.

 

Specifically the `<origin>` tag within the `<joint>` tag changes.

 

Is this intended?
If yes, how can I avoid this behaviour?
If no, will you resolve this or is there a known work around?

 

Kind regards,
Thomas

Gregwar, 2021-03-16:
Hello,

I am not sure to understand what your problem is exactly
Can you explain with more details?
Like what you expect compared to what you actually get ?

Joints should be located exactly where you see them in Onshape, as explained here:
https://onshape-to-robot.readthedocs.io/en/latest/design.html#joint-frames

Feel free to give me the assembly with problems, or a simple proof-of-concept reproducing your problem (gregoire.passault@u-bordeaux.fr)

PS: I am not related to the Onshape team, this is an independent project using Onshape API

ThomasPiechulek, 2021-03-19:
Hello,
okay we thought about an answer to this because the matter is quite difficult to describe.
So we copied an assembly and loaded it with onshape-to-robot-bullet. Than we observerd jitters in many joints and faulty Origin and rpy values, as we checked the urdf.

`
<link name="upper_part_yang">
<visual>
<origin xyz="-1.73472e-17 0 -0.005" rpy="1.5708 -8.47206e-18 1.0957e-18" />
<geometry>
<mesh filename="package://upper_part_yang.stl"/>
</geometry>
<material name="upper_part_yang_material">
<color rgba="0.901961 0.901961 0.901961 1.0"/>
</material>
</visual>
<collision>
<origin xyz="-1.73472e-17 0 -0.005" rpy="1.5708 -8.47206e-18 1.0957e-18" />

<geometry>
<mesh filename="package://upper_part_yang.stl"/>
</geometry>
<material name="upper_part_yang_material">
<color rgba="0.901961 0.901961 0.901961 1.0"/>
</material>
</collision>
<inertial>
<origin xyz="3.57735e-06 -0.0268018 -0.0429114" rpy="0 0 0"/>
<mass value="0.0595083" />
<inertia ixx="6.0336e-05" ixy="1.74995e-09"  ixz="-1.27973e-09" iyy="0.000104812" iyz="-7.98351e-07" izz="6.47565e-05" />
</inertial>
</link>
`
is changed to :
`<link name="upper_part_yang">
<visual>
<origin xyz="2.77556e-17 0 -0.005" rpy="1.5708 -1.15036e-30 1.06624e-17" />
<geometry>
<mesh filename="package://upper_part_yang.stl"/>
</geometry>
<material name="upper_part_yang_material">
<color rgba="0.901961 0.901961 0.901961 1.0"/>
</material>
</visual>
<collision>
<origin xyz="2.77556e-17 0 -0.005" rpy="1.5708 -1.15036e-30 1.06624e-17" />
<geometry>
<mesh filename="package://upper_part_yang.stl"/>
</geometry>
<material name="upper_part_yang_material">
<color rgba="0.901961 0.901961 0.901961 1.0"/>
</material>
</collision>
<inertial>
<origin xyz="3.57735e-06 -0.0268018 -0.0429114" rpy="0 0 0"/>
<mass value="0.0595083" />
<inertia ixx="6.0336e-05" ixy="1.74995e-09"  ixz="-1.27973e-09" iyy="0.000104812" iyz="-7.98351e-07" izz="6.47565e-05" />
</inertial>
</link>`

Thank you for your time.
Regards 
Thomas

closed 2025-02-20 by Gregwar