Rhoban/onshape-to-robot #50: Issue reading limits

raviBhadeshiya, 2021-04-01:
Got this error while reading Limits for floating joint

Traceback (most recent call last):
  File "/home/onshape-to-robot/onshape-to-robot", line 3, in <module>
    from onshape_to_robot import onshape_to_robot
  File "/home/onshape-to-robot/onshape_to_robot/onshape_to_robot.py", line 15, in <module>
    from .load_robot import \
  File "/home/onshape-to-robot/onshape_to_robot/load_robot.py", line 166, in <module>
    limits = getLimits(jointType, data['name'])
  File "/home/onshape-to-robot/onshape_to_robot/features.py", line 84, in getLimits
    minimum = readParameterValue(parameter)
  File "/home/onshape-to-robot/onshape_to_robot/features.py", line 61, in readParameterValue
    if value['message']['booleanValue'] == parameterValue:
KeyError: 'booleanValue'

Do you guys know any workaround?

closed 2021-04-01 by raviBhadeshiya

Gregwar, 2021-04-02:
Hello, why did you close this? Did you solve your issue?

raviBhadeshiya, 2021-04-02:
Ended up finding a mistake on Onshape Side!