Rhoban/onshape-to-robot #55: Cannot Retrieve Workspace ID

RX-00, 2021-06-28:
Hello! I've been trying to use this useful program but I seem to be having trouble actually retrieving my OnShape Workspace. I have followed the documentation and created a new API key on OnShape of which I've put into my ~/.bashrc and sourced. I have deleted and created multiple keys but I always end up with the same result:

>>>$ onshape-to-robot onshape-to-robot-examples/2wheels/

* Checking OpenSCAD presence...
* Checking MeshLab presence...

* Retrieving workspace ID ...
https://cad.onshape.com/api/documents/862948a6ea6d38343e1d3272?
! ERROR (403) while using OnShape API

Gregwar, 2021-06-28:
Hello,

This issue is explained here:
https://github.com/rhoban/onshape-to-robot-examples/#why-do-i-get-error-403-while-using-onshape-api-

The issue is that you can't access the document with API because it doesn't belongs to you, unfortunately you have to make a copy and use the copy instead if you want to try to re-import the examples yourself

RX-00, 2021-06-29:
Thank you for the reply! Unfortunately I also get the 403 error on the OnShape documents that I do own.

$ onshape-to-robot /my-robot
```
* Checking OpenSCAD presence...
* Checking MeshLab presence...

* Retrieving workspace ID ...
https://cad.onshape.com/api/documents/3758cd5dea39008a96c66096?
! ERROR (403) while using OnShape API
```

This is my project that I'm trying to convert: https://cad.onshape.com/documents/3758cd5dea39008a96c66096/w/2ee96f77150708f4482d2d4b/e/ca99fdba740bd6d4735df501

I was able to successfully do this with another computer a couple months back. Do I need to delete my Oauth on OnShape and make a new one?

Gregwar, 2021-07-01:
Can you check that the clock of your computer is in sync with international time ?
I figured out that it matters for API challenges to check out

RX-00, 2021-07-11:
That was the solution! I would have never realized that haha. Thank you so much!

closed 2021-07-11 by RX-00

Gregwar, 2021-07-12:
I personally struggled a while about that one already as well :slightly_smiling_face: