Rhoban/onshape-to-robot #59: Ignoring configuration option in config.json

agillies8, 2022-03-07:
When I use the export tool, it seems to ignore the configuration specification in the config.json file:

Eg:

`
"documentId": "abcdefghi123456789",
"outputFormat": "urdf", 
"assemblyName": "myrobot",
"versionId": "abcdefghi123456789",
"configuration: "extra long legs",
`

no matter what I put, the default configuration is always exported. Is there something I am missing?

Additionally - some of my links in the top level assembly are sub-assemblies. Is there any way to specify configurations for these sub-assemblies?

Gregwar, 2022-03-08:
Hello,

# Setting the "configuration name"

I guess you are using a configuration "list" to set your global configuration. When you do that, OnShape generates a random name for the configuration variable you can't see. Here are the steps if you want to use it from the API:

Go to your assembly, your URL should looks like:

```
https://cad.onshape.com/documents/XXX/w/YYY/e/ZZZ
```

Replace `documents` with `api/assemblies/d/`, it should look like:

```
https://cad.onshape.com/api/assemblies/d/XXX/w/YYY/e/ZZZ
```

You will see something similar to:

```
...
"fullConfiguration" : "List_I4JxJYIGTR44Gf=Default"
...
```

Here, `List_I4JxJYIGTR44Gf` is what you are searching for. You can now set something like:

```
"configuration": "List_I4JxJYIGTR44Gf=extra long legs"
```

This should be fixed in the future to use the `/api/elements/d/.../configuration` API to retrieve the actual names of the parameters, so that we could write `"configuration": "Configuration=Something"` instead.

# Forwarding configuration to sub-assemblies

You can use the configuration of your top-level assemblies to define which configuration will be used for a sub-assembly. This can be done by clicking on "Configure assembly features" on the top on your list, then click on the part on your tree, it will add the possibility to configure which instance will be used depending on the top-level assembly configuration.

Hope it helps

agillies8, 2022-03-10:
Thanks for this @Gregwar for the rapid and thorough response.  I was actually using configuration tables, not list input - but I went back and changed to a list input and was able to find the List ID you mentioned. Unfortunately it still seems to ignore the top level configuration settings.

Additionally, one of my sub assemblies is still being included in the urdf export despite seemingly any combination of configuration and suppression settings I am applying. This may be a separate issue, but including here in case they happen to be related?

Gregwar, 2025-02-20:
This is now done, configuration can be set using names, including lists.

For example:

[image]

Can be described as `Configuration=Long;RemovePart=true;Length=30mm` in the configuration

See
http://localhost:8080/config.html#configuration-default-default

hshi74, 2025-05-12:
Hi, thanks for this fantastic tool! I ran into an issue while exporting an assembly with configurations:

This is my onshape assembly:
[image]

This is my config.json
`
{'document_id': '3084b13ad43394bd46cc00cf', 'output_format': 'mujoco', 'robot_name': 'left_leg_XC430', 'assembly_name': 'leg_XC430', 'configuration': 'Configuration=Left', 'include_configuration_suffix': False, 'draw_frames': True, 'merge_stls': True, 'simplify_stls': True, 'maxSTLSize': 1}`

Somehow I received this error message:
```
https://cad.onshape.com/api/assemblies/d/3084b13ad43394bd46cc00cf/w/bfec0c698465027acb50f9f3/e/ffd0b5823bacb55a731d832c?includeMateFeatures=true&includeMateConnectors=true&includeNonSolids=true&configuration=List_Sp2Yhk1niQQ3vQ%3DLeft
! ERROR (400) while using Onshape API
! {
  "moreInfoUrl" : "",
  "message" : "Invalid configuration parameter value: '\"Left\"as List_Sp2Yhk1niQQ3vQ_conf' for parameter id: List_Sp2Yhk1niQQ3vQ",
  "status" : 400,
  "code" : 9999
}
```

Did I miss something obvious? Would really appreciate some input!

Gregwar, 2025-05-12:
After some tests, it appear that the behaviour you mention comes from Onshape, where the first item can only be called "Default"

I will open a support ticket with them to clarify. In the meantime I suggest as a workaround to keep a first "Default" option and add your own below

Gregwar, 2025-05-12:
It seems that, in the current state of Onshape, `Default` should be used for the first item of an enum, even if you rename the item. I need to know whether this is the intended behaviour or if it's a bug.

I opened an Onshape support ticket.

In the meantime, I pushed a workaround on `1.6.2`, that effectively use `Default` as a value when the configuration is set to the first item of an enum

hshi74, 2025-05-12:
Amazing! Thanks for the prompt response. The workaround works for me.

Gregwar, 2025-05-12:
Great!

I'll keep this issue open until I get a response from Onshape support

Gregwar, 2025-05-13:
Just got news from support. Actually, the very first input made by the user in the box remains the value that should be used when passed as a configuration string to Onshape.

This value is available under another key when retrieving the configuration parameters (below an example). I just updated the code accordingly, `1.6.3` should now address properly this issue.

```js
// Example of output for element/.../configuration 
...
"options": [
    {
      "btType": "BTMEnumOption-592",
      "optionName": "Normal",
      "option": "Default",
      "nodeId": "M1gZTJQjWWP7m55Rr"
    },
    {
      "btType": "BTMEnumOption-592",
      "optionName": "Long",
      "option": "MyFirstEntry",
      "nodeId": "Mdobxc1mp+H4nPcv3"
    }
 ],
```

closed 2025-05-13 by Gregwar