Rhoban/onshape-to-robot #63: Error 406 while using OnShape API: Request not acceptable

michaellin, 2022-04-26:
It seems like something has changed on the OnShape API side that is breaking even the basic URDF or SDF exporting. Full message is:

! ERROR (406) while using OnShape API
! {
  "moreInfoUrl" : "",
  "message" : "Request not acceptable.",
  "status" : 406,
  "code" : 0
}

Gregwar, 2022-04-26:
Hello
Sorry I am away, will be back next week and check that then
If you have some time you can check which underlying request fails

Gregwar, 2022-04-27:
Apparently this comes from the `Accept` header that was too restrictive, should be fixed now

davidschnermann, 2022-04-28:
Hello Gregwar,
thanks a lot for your fast fix and your work in general!
It seems to work.
The only thing is, that the program fails to exit correctly. 
To me it ends with:
```
* Writing URDF file
Traceback (most recent call last):
  File "/home/david/anaconda3/envs/Onshape/bin/onshape-to-robot", line 8, in <module>
    sys.exit(onshape_to_robot())
TypeError: 'module' object is not callable
```

Gregwar, 2022-04-30:
@davidschnermann what version are you using?

The `bin/` directory isn't a thing since https://github.com/Rhoban/onshape-to-robot/pull/60/files

If you installed it with `pip` it should work, else (if you cloned the repository) you can also consider running directly the  module:

```
python -m onshape_to_robot.onshape_to_robot
```

closed 2025-02-20 by Gregwar