Rhoban/onshape-to-robot #74: Support for obj format

dpetrillo740, 2022-05-29:
I would like to use onshape-to-robot to load robots into Isaac Sim. Currently Isaac Sim only allows meshes in obj format. Can there be an option to use OBJ format instead of stl for exported meshes?

dpetrillo740, 2022-05-30:
I confirmed that if I generate the urdf file using onshape-to-robot and then manually replace the references to the individual .stl files with references to .obj files that I export manually from onshape, that it can then be uploaded and used in Isaac Gym effectively. Maybe the other features for combining the .stl visual and geometry, etc will not work. It would still be nice to have this option to make it functional for Isaac Gym

bpousett, 2025-05-19:
Regarding the export to .obj, one good workaround solution is just to do this conversion in batch with MeshLab: https://sourceforge.net/p/meshlab/discussion/499533/thread/51e3e2c0/.

Running the following bash script in the folder with all the meshes worked well for me:
```
for file in *.stl; do
    meshlabserver -i "$file" -o "${file%.*}.obj" # -om vc
done
```
This is much quicker than using onshape to export .obj manually.

Gregwar, 2025-05-20:
Thanks @bpousett for the suggestion
However,  I think obj files have the advantage of supporting other features (especially colors and textures), that would be lost in the process

TheKing349, 2026-09-30:
Hello! I would also like to use obj format in onshape-to-robot. The meshlab script does work but adds complexity to the export process. OBJ is a largely supported mesh format and would be nice to have 1st party support with it.