Rhoban/onshape-to-robot #77: Joint limits searched for in wrong assembly

rsinnet, 2022-07-04:
During initial feature loading, a request is made to the Onshape API, but this request can be on the wrong assembly. It seems the request is made on what becomes the root assembly, but that can be the first assembly in the assembly tab. The actual root assembly has the same name as the tab.

I fixed this in our fork, but reporting it here FYI.

Gregwar, 2022-07-04:
Hello Ryan,

Thanks for your PRs!
I will likely not be able to review them anytime soon since I am currently preparing for RoboCup 2022 that will be held next week

Do not hesitate to ping be again mid-july!

Gregwar, 2025-02-20:
Many things changed in the codebase, I believe this would work now
Do not hesitate to re-open else

closed 2025-02-20 by Gregwar