Rhoban/onshape-to-robot #84: Bodyless Frames

Jarris81, 2022-08-18:
Hi everyone,

Adding a small box to add frames to a robot seems a little redundant. Is it possible to simply use a **Mate Connector** as a frame for an robot/object? The following picture should give an example, in which the **pos1** suppose to be a frame and a child link in the resulting .urdf file.
[image]

 I would try to add this feature myself, but maybe someone can show me where to look in the code. I'm totally new to this repo.

Thank you

Gregwar, 2022-08-18:
Hello
Thanks for suggestion, it is true that it would be simplier that way

However, would it be possible then to place some mate at a position with offsets or other constraints?
For instance 3cm away from a surface etc.

Currently it can be done with bodies, the first constraint must be with the body to which the frame is attached to but extra constraints can be used to position the frame

Jarris81, 2022-08-18:
Hello,

thanks for the reply. I agree, setting a frame with a specific offset is not possible with a mate connector. Maybe a good solution to fulfill both requirements would be to simply make each frame a bodyless link in the urdf?

I tried to use the frame option just now, but I get the following error:

`  File "/home/jason/.pyenv/versions/3.10.0/envs/basf/lib/python3.10/site-packages/onshape_to_robot/load_robot.py", line 300, in <module>
    frames[occurrenceB].append(
KeyError: 'MJyxqs0WHD4O/PPJ/'`

Any idea what could be the reason for this?

Thanks

Gregwar, 2022-08-18:
Frames are already made bodyless when imported in urdf, it is only a body in onshape

Did you click the objects in proper order (child first then parent) in the frame constraint?

Else, you can share the document with me, but I might be unable to help before a few days since I am not at home now

Gregwar, 2022-08-18:
(if `draw_frames` option is set to true in the config json, frames bodies are kept)

Jarris81, 2022-08-19:
I  tried both orders, but no luck at. The error keeps coming up. 

Here is the link to the document: https://cad.onshape.com/documents/c6bbee4c679f1136517dbd13/w/93d55c9c58e5f3dcd5e8982d/e/443ae538899342843556ddb6?renderMode=0&uiState=62ff5431d3251e0c4486d78c

Thanks in advance!

Gregwar, 2022-11-05:
Hello,
Sorry for not giving any news sooner, can you share access to your document so that I can make a copy to run the import?

closed 2025-02-20 by Gregwar