Rhoban/onshape-to-robot #86: Loops in links

mfrimannm, 2022-09-11:
Hi

I have a problem I don't know how to solve.
In this model
[https://cad.onshape.com/documents/c3f44096b0f5e4646ed48e3f/w/670467f68cac02b087631e3d/e/aa91522f7b6623958d10d37b?renderMode=0&uiState=631e3a4c38ae3c7ec87ec8db](https://cad.onshape.com/documents/c3f44096b0f5e4646ed48e3f/w/670467f68cac02b087631e3d/e/aa91522f7b6623958d10d37b?renderMode=0&uiState=631e3a4c38ae3c7ec87ec8db)

I have two links going to the same block, but it seems like the software doesn't like it says it is already connected.

What can I do?

Gregwar, 2022-11-05:
Hello,
There can't be loops in produced models currently.

However, as far as I understand your design, no loop seems needed, you could simply use a DoF only on one side of your mechanism, can't you?

Gregwar, 2025-02-19:
Kinematics loops can now be handled, see: 
https://onshape-to-robot.readthedocs.io/en/latest/kinematic_loops.html

closed 2025-02-19 by Gregwar