Rhoban/onshape-to-robot #88: Suppressed parts in Onshape version are not ignored by onshape-to-robot command

maxwellkm, 2022-11-02:
TL;DR: Expected behavior is that suppressed parts in Onshape should not be encountered by onshape-to-robot

Vendors will often only provide surface models for components, rather than a solid model, in an effort to protect IP. We have no option but to use these models in our CAD for certain components. It seems onshape-to-robot has difficulty and will fail to execute when it encounters such models, which may have varying degrees of "validity" (e.g., non-watertight, discontinuous, etc)

As a user, I'd expect that simply suppressing these problem surface models in the assembly prior to creating a version for URDF export should be enough to keep them away from onshape-to-robot. In practice, the function still attempts to load these models, and ultimately fails, preventing the rest of the URDF from being generated.

My only workaround for this has been to create a new branch of the assembly, in which I explicitly delete all surface models (and traces of them) from the assembly, and then export the URDF from there. This has several obvious issues: 1) these are important components and really do belong in the assembly; 2) it may not always be possible to delete these models if they are critical for mates involved in certain joints; 3) it's an overall clunky and human error prone workflow.

Gregwar, 2022-11-05:
Hello, suppressed items should indeed not be loaded.
Can you make the smallest possible example reproducing the problem and share the output and the Onshape document ?

Gregwar, 2025-02-20:
I just tried the current codebase on @maxwellkm example
Both visual and mass seems to update properly (The codebase changed a lot recently)

I will close this, do not hesitate to re-open if I got you wrong

## Unsupressed

[image]
[image]

## Suppressed

[image]
[image]

closed 2025-02-20 by Gregwar