Rhoban/onshape-to-robot #89: ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)

andrestoga, 2022-11-04:
Hi,

I'm getting the following error. Any help would be appreciate it.

```
(onshape_robot) kip@MX2413NB:~/onshape_robot$ onshape-to-robot robot
pybullet build time: Nov  4 2022 16:45:40
* Checking OpenSCAD presence...
Can't run openscad -v, disabling OpenSCAD support
TIP: consider installing openscad:
sudo add-apt-repository ppa:openscad/releases
sudo apt-get update
sudo apt-get install openscad
* Checking MeshLab presence...

* Retrieving workspace ID ...
Traceback (most recent call last):
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/urllib3/connectionpool.py", line 386, in _make_request
    self._validate_conn(conn)
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/urllib3/connectionpool.py", line 1042, in _validate_conn
    conn.connect()
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/urllib3/connection.py", line 414, in connect
    self.sock = ssl_wrap_socket(
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/urllib3/util/ssl_.py", line 449, in ssl_wrap_socket
    ssl_sock = _ssl_wrap_socket_impl(
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/urllib3/util/ssl_.py", line 493, in _ssl_wrap_socket_impl
    return ssl_context.wrap_socket(sock, server_hostname=server_hostname)
  File "/usr/lib/python3.10/ssl.py", line 513, in wrap_socket
    return self.sslsocket_class._create(
  File "/usr/lib/python3.10/ssl.py", line 1071, in _create
    self.do_handshake()
  File "/usr/lib/python3.10/ssl.py", line 1342, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/requests/adapters.py", line 489, in send
    resp = conn.urlopen(
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    retries = retries.increment(
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='cad.onshape.com', port=443): Max retries exceeded with url: /api/documents/a0881d6afc6bcafd94cd037b (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)')))
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/kip/venv/onshape_robot/bin/onshape-to-robot", line 8, in <module>
    sys.exit(onshape_to_robot.main())
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/onshape_to_robot/onshape_to_robot.py", line 17, in main
    from .load_robot import \
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/onshape_to_robot/load_robot.py", line 25, in <module>
    document = client.get_document(config['documentId']).json()
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/onshape_to_robot/onshape_api/client.py", line 122, in get_document
    return self._api.request('get', '/api/documents/' + did)
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/onshape_to_robot/onshape_api/onshape.py", line 209, in request
    res = requests.request(method, url, headers=req_headers, data=body, allow_redirects=False, stream=True)
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/requests/sessions.py", line 587, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/requests/sessions.py", line 701, in send
    r = adapter.send(request, **kwargs)
  File "/home/kip/venv/onshape_robot/lib/python3.10/site-packages/requests/adapters.py", line 563, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='cad.onshape.com', port=443): Max retries exceeded with url: /api/documents/a0881d6afc6bcafd94cd037b (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)')))
```

Gregwar, 2022-11-05:
Hello,

Can you specify your OS?

If you are on MAC, there is this StackOverflow issue you can check on:
https://stackoverflow.com/questions/52805115/certificate-verify-failed-unable-to-get-local-issuer-certificate

andrestoga, 2022-11-07:
I've had this issue using WSL with Ubuntu 22.04 and VMware with Ubuntu 22.04

Wonder if the solution you posted can be applied to Ubuntu.

Does onshape-to-robot works on Windows 10?

Gregwar, 2022-11-07:
I've never tested it on Windows. Some features like mesh simplification and pure shape requires external programs and I am not sure about them. But I think most of it should be able to run since it's only Python.

Can you try `pip install pip install pip-system-certs` ?

andrestoga, 2022-11-11:
I did and I'm having the same error....

closed 2025-02-20 by Gregwar