Rhoban/onshape-to-robot #94: Robot with multiple joints on one link

BrianKSchultz, 2022-12-08:
Hello there, thanks for this great tool. I'm trying to create a URDF for an [ABB IRB8700 robot](https://new.abb.com/products/robotics/industrial-robots/irb-8700). This model has a few extra linkages compared to a normal 6 dof robot arm, due to the counterweight and linear actuators for link2. If I suppress the counterweight, link, and linear actuators along with their associated mates/joints, the script works perfectly. With everything active, I get an error about connecting to multiple child links (e.g. Link3 is a child of both Link2 AND the counterweight link).

Do you have a suggestion for the best way to handle this? Thanks for your help. - Brian

[image]

Gregwar, 2022-12-09:
Hello,

We dont handle kinematics loop currently. I think that it cant be described using UFDF unfortunately 

One solution is to open the kinematic chain and use software to maintain the closure constraints true

danem, 2023-02-23:
Just for clarification (as a non mech-e), does this limitation apply to something like Ackermann steering? I'm trying to model this, and I'm getting the following error:
```
Error, the relation ack_bar_right is connected a child that is already connected
Be sure you ordered properly your relations
```
[image]

Thanks

Gregwar, 2023-02-23:
Hello,

You can have multiple joint on one link, but the kinematic structure should be a tree currently. This is a current limitation of this tool.

The reason is that URDF doesnt support closed loops. SDF does though, so that would be interresting to have it.

Current workaround is to make your robot kinematics a tree and solving the closing of the chain manually on software 

What is the purpose you have in mind?

atticusrussell, 2023-02-23:
I am currently encountering a similar issue with a rudder. Do you have any suggestions?
[image]

```* Getting assembly features, scanning for DOFs...
WARNING: joint dof_servo_arm1 of type revolute has no limits
+ Found DOF: servo_arm1 (revolute)
WARNING: joint dof_arm1_arm2 of type revolute has no limits
+ Found DOF: arm1_arm2 (revolute)
WARNING: joint dof_arm2_rudder of type revolute has no limits
+ Found DOF: arm2_rudder (revolute)
WARNING: joint dof_rudder_base of type revolute has no limits
+ Found DOF: rudder_base (revolute)

Error, the relation rudder_base is connected a child that is already connected
Be sure you ordered properly your relations, see:
https://onshape-to-robot.readthedocs.io/en/latest/design.html#specifying-degrees-of-freedom
```

Gregwar, 2023-02-24:
Just to be sure about your use cases so that I can write tutorials for them:

* Is your goal to simulate the robots? Which simulator?
* is your goal to use the description format? With what purpose?

atticusrussell, 2023-02-24:
My goal is to use this tool to create URDF to simulate in Gazebo with ROS2. I wish to simulate the physical aspects of the robot so I can write control code for the electronics before fabricating the actual robot.

I have also been having some trouble getting an exported robot working with ROS2/Gazebo, even in a design without this closed kinematics error, so a tutorial/guidance on that would also be much appreciated.

Sambertibus, 2024-04-25:
Hi, I've encountered similar issues. I discovered that you can use PyBullet to simulate this using constraints. Check out page 17 of the quick start guide for more information: https://dirkmittler.homeip.net/blend4web_ce/uranium/bullet/docs/pybullet_quickstartguide.pdf.

Gregwar, 2024-04-29:
Indeed, to handle loops in kinematics chain, you need to add runtime constraints.

When designing the robot, you can add mates with the name `closing_something` (prefixed with `closing_`). Onshape-to-robot will add a frame on each side
See: https://onshape-to-robot.readthedocs.io/en/latest/design.html#closing-frames

You can then use pyBullet's `createConstraint` to add the constraint

If you think that would be relevant, I could write more documentation and provide some end-to-end example for this

minched, 2024-06-01:
Plus one. I would be happy to see just a bit more documentation about it or a video showing exactly this. 
Thanks.

Gregwar, 2024-06-03:
I added a dedicated page on the documentation:
https://onshape-to-robot.readthedocs.io/en/latest/kinematic_loops.html

Does it clarify on how it works ?

minched, 2024-06-04:
Yes, this adds defiinitly the clarity I needed - especially the Onshape example.  Working now as well. 
And Placo is new to me, but looks interesting as well! I will dig into that a bit more. Thanks!

Gregwar, 2024-06-04:
> Yes, this adds defiinitly the clarity I needed - especially the Onshape example. Working now as well.
Great!

> And Placo is new to me, but looks interesting as well! I will dig into that a bit more. Thanks!
Placo is the library we developped in my team, it is written in C++ with Python bindings and offers a task-space inverse kinematics API, do not hesitate to ping me for any question about it

closed 2024-06-04 by Gregwar