Rhoban/onshape-to-robot #96: Very big robot apear the vanish in pybullet

FrGe2016, 2023-02-01:
Hi

I have tried a very big robot basically a gantry crane with a few arms ( 30m X 10 m X 10m ) moving along a 60m rail.
It apears in bullet for about 1 second then vanishes. 

If i use ros2
The robot is displayed in rviz and every joint move perfectly 
But i also have simulation problems with gazebo. ( the robot definition is available but no display )

Gregwar, 2023-02-01:
Hello,

Do you pass `-f` to `onshape-to-robot-bullet` ?
This makes the robot "fixed" (else, it is considered as floating base, like a mobile robot)

https://onshape-to-robot.readthedocs.io/en/latest/commands.html#onshape-to-robot-bullet-running-bullet-simulation-to-test-your-robot-urdf

Gregwar, 2025-02-20:
Closing this because of lack of activity
Don't hesitate to re-open if you have any news

closed 2025-02-20 by Gregwar