Rhoban/onshape-to-robot #100: KeyError: 'suppressed'

JHPHELAN, 2023-04-11:
Recently installed OnShape-to-robot plus accessories on a Dell laptop / Ubuntu 22.04  successfully.
my config.json points with appropriate access and secret keys to the JPL Open Source Rover:
[https://cad.onshape.com/documents/5c358484057fc4c01d5e5803/w/bdf583acdf29d75695be5fbc/e/e3eb7193ecdec5b0a710978b](url)
`ubuntu@LinuxBox:~$ onshape-to-robot OnShape-OSR`
```pybullet build time: Apr  9 2023 19:54:23
* Checking OpenSCAD presence...
* Checking MeshLab presence...

* Retrieving workspace ID ...
+ Using workspace id: ec57fc1702d5177d685a86c9

* Retrieving elements in the document, searching for the assembly...
+ Found assembly, id: ccf42f743a831a145aea7c8a, name: "Rover"

* Retrieving assembly "Rover" with id ccf42f743a831a145aea7c8a

* Getting assembly features, scanning for DOFs...
* Found total 0 DOFs

* Building robot tree
* Trunk is Mechanical Integration <1>
Traceback (most recent call last):
  File "/home/ubuntu/.local/bin/onshape-to-robot", line 8, in <module>
    sys.exit(onshape_to_robot.main())
  File "/home/ubuntu/.local/lib/python3.10/site-packages/onshape_to_robot/onshape_to_robot.py", line 17, in main
    from .load_robot import \
  File "/home/ubuntu/.local/lib/python3.10/site-packages/onshape_to_robot/load_robot.py", line 336, in <module>
    if not occurrence['instance']['suppressed']:
**KeyError: 'suppressed'`**
```
The 'KeyError: 'suppressed' error apparently refers to 'suppressed' not being in the 'defaultdict' referred to in 'load_robot.py'.
I can't find the actual dictionary despite considerable digging.
I'll see what happens if I 'comment out' that block of code....

JHPHELAN, 2023-04-14:
@Gregwar 
It seem from what I gather it's not that we're not testing correctly for 'suppressed' that 'suppressed' needs to be added to some dictionary.

Gregwar, 2025-02-20:
Hello,
Many changes in the codebase recently that are likely to change the behavior
If you do more tests and want to provide some feedbacks on the current codebase, feel free to re-open

closed 2025-02-20 by Gregwar