Rhoban/onshape-to-robot #102: Wrong mass computation of assemblies with standard components

dettmann, 2023-06-14:
Hi,

when I create an assembly from own parts and parts coming from onshape's standard component catalogue (e.g. a hex nut), I noticed that the mass computation is not correct. This is because the _massProperties_ (in line 143 of onshape_to_robot.py) show that all standard components have zero mass. The weired thing is that within onshape the masses are set correctly and also the computation of an assembly with standard components is correct. So, is this a bug somewhere in onshape-to-robot or directly in the onshape API?

I appreciate any thoughts, thanks.

csamid, 2023-08-21:
Late to the party, but this error occurs because Onshape's standard content is not considered an Onshape part :( 
However, there is a work around. Export the standard content as a STEP file and then import that STEP file into the document :)

Niklas-Mulsow, 2023-08-23:
Hi together, 

as i understand that right, this would mean we have always to replace every screw, washer, nut, etc. by hand via the step ex- and import? Since it is part of our business to create simulation models for robotics, this greatly diminishes the use of onshape for us. Can you please verify, if there is a better option available, for example to do the replacement automatically with a sricpt. 

Best 
Niklas

csamid, 2023-08-24:
Hi Niklas,

It might be possible to automate this process with [Onshape's Model Translation API](https://onshape-public.github.io/docs/api-adv/translation/) I can look into implementing this if you'd like. If so, can you provide me with a simplified model to test on. Finally, on average how many unique standard content parts are used per robot model?  

Best
Samid

csamid, 2023-08-29:
Hi Niklas,

Good news. I think I have a working solution. I created a very simple model, so no rush on your part. I'll create a pull request later this week. 

Best
Samid

Niklas-Mulsow, 2023-08-31:
Hello, sounds good, Thank you very much.

Best, Niklas

Niklas-Mulsow, 2023-09-05:
Hey, 

i got a message from the onshaoe support to that issue: Please have a look on it:

The issue is with the logic in the api call.  They are calling for mass properties of each part.  Part metadata responses show whether the part is Standard Content or not.  What the author of the scipt is missing is when this call is used for standard content parts, they need to pass the document id that the standard content is in in the linkedDoucmentId field.  Then you will get the mass.
 
I did a specific test of this and saw 0 mass without it and XYZ of mass with it.  Response example:
 
Without linkedDocumentId
{
  "bodies": {
    "JHD": {
      "principalInertia": [
        0,
        0,
        0
      ],
      "principalAxes": [
        {
          "btType": "BTVector3d-389",
          "z": 0,
          "x": 1,
          "y": 0
        },
        {
          "btType": "BTVector3d-389",
          "z": 0,
          "x": 0,
          "y": 1
        },
        {
          "btType": "BTVector3d-389",
          "z": 1,
          "x": 0,
          "y": 0
        }
      ],
      "mass": [
        0,
        -1.1873558866243225e-10,
        1.1873558866243225e-10
      ],
      "volume": [
        0.0000012938125651453984,
        0.0000012936935647496897,
        0.0000012939315655411076
      ],
      "periphery": [
        0.0009838007856934833,
        0.0009837897816683328,
        0.0009838117897186338
      ],
      "centroid": [
        0,
        0,
        0,
        -0.000001,
        -0.000001,
        -0.000001,
        0.000001,
        0.000001,
        0.000001
      ],
      "inertia": [
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        -1.2938125651453983e-12,
        -1.2938125651453983e-12,
        -1.2938125651453983e-12,
        -1.2938125651453983e-12,
        -1.2938125651453983e-12,
        -1.2938125651453983e-12,
        -1.2938125651453983e-12,
        -1.2938125651453983e-12,
        -1.2938125651453983e-12,
        1.2938125651453983e-12,
        1.2938125651453983e-12,
        1.2938125651453983e-12,
        1.2938125651453983e-12,
        1.2938125651453983e-12,
        1.2938125651453983e-12,
        1.2938125651453983e-12,
        1.2938125651453983e-12,
        1.2938125651453983e-12
      ],
      "hasMass": false,
      "massMissingCount": 1
    }
  },
  "microversionId": "93a5e600bbcd93bdb7bd044d"
}
 
With linkedDocumentId
 
{
  "bodies": {
    "JHD": {
      "principalInertia": [
        6.443038617919366e-8,
        0.0000014109841454990842,
        0.0000014135436698420994
      ],
      "principalAxes": [
        {
          "z": 1,
          "x": 0,
          "y": 0
        },
        {
          "z": 0,
          "x": 1,
          "y": 0
        },
        {
          "z": 0,
          "x": 0,
          "y": 1
        }
      ],
      "mass": [
        0.010389314898117548,
        0.010389312652981377,
        0.010389317143253722
      ],
      "volume": [
        0.0000012938125651453982,
        0.0000012936935647496895,
        0.0000012939315655411074
      ],
      "periphery": [
        0.0009838007856934833,
        0.0009837897816683328,
        0.0009838117897186338
      ],
      "centroid": [
        1.2536974258291559e-18,
        7.135062707114396e-18,
        -0.017776927663227745,
        -0.0000010001804666733743,
        -0.0000010002912075783782,
        -0.01777793742517876,
        0.000001000180466675841,
        0.000001000291207592651,
        -0.017775917901280722
      ],
      "inertia": [
        0.0000014109841454990842,
        0,
        0,
        0,
        0.0000014135436698420994,
        0,
        0,
        0,
        6.443038617919366e-8,
        0.0000014109825156227593,
        -1.297141190798301e-12,
        -1.3184106310348743e-12,
        -1.297141190798301e-12,
        0.0000014135420432070719,
        -1.3334637074334925e-12,
        -1.3184106310348743e-12,
        -1.3334637074334925e-12,
        6.442896833561772e-8,
        0.0000014109857753752806,
        1.297141190836796e-12,
        1.318410630643441e-12,
        1.297141190836796e-12,
        0.0000014135452964769984,
        1.333463705853549e-12,
        1.318410630643441e-12,
        1.333463705853549e-12,
        6.443180402276963e-8
      ],
      "hasMass": true,
      "massMissingCount": 0
    }
  },
  "microversionId": "93a5e600bbcd93bdb7bd044d"
}

Best to inform the author they need to check if the part metadata comes back as SC however you can pass that same id for all the parts if you want and it would solve this.
Lou Gallo
PD/UX/Support/Community
Onshape, Inc.

csamid, 2023-09-05:
Oh great, this is a much better than the solution I came up with. Will try to implement this instead.

Thanks,
Samid

Niklas-Mulsow, 2023-09-06:
Perfecto, 

thank you a lot.

Best Niklas

csamid, 2023-09-07:
Oh yea, that could be a problem. Should I include this in the #110 pr? I would just modify the `get_mass_properties` method to include `linkDoucmentId` parameter.

minched, 2024-06-01:
Superb! Thanks for onshape-to-robot and thanks for fixing this issue already. I just noticed the same flaw and have the same issue of a multitude of bolts. 
Is there a reason the pull-request is still not answered and pushed to the master? 
Thanks again!

Gregwar, 2024-06-03:
Thanks for upping @minched 
I merged PR #110 and pushed `v0.3.26` (on pip)

Can you give it a try and tell me if it is OK now?

PR #110 (2024-06-03): Add standard content mass properties retrieval
### Summary
Adds `standard_cont_mass_properties` method to the `Client` class, enabling mass properties retrieval for standard content. 

### Testing
Testing was done using this [onshape document](https://cad.onshape.com/documents/017f1c2072e843891025e97d/w/3d653d232bb5b46cbdfc21c6/e/45b2f6a5cd97e5c57fe8e5a6)
After the changes the inertial tag for `sub-asm-std-cont` matches with the `display mass properties` window within onshape. 

Inertial tag:
```urdf
<inertial>
<origin xyz="3.9625718201122967351e-10 7.420351968363106984e-05 0.09976964273767550 [... cut]
closed 2024-06-03 by Gregwar