Rhoban/onshape-to-robot #103: help selecting configuration

atticusrussell, 2023-07-04:
I am trying to export a simplified configuration ("collisions") in my onshape assembly, however when I go to export the robot, the default configuration seems to always be chosen. I found this old issue #59 that seems extremely similar, but following the advice about replacing `documents` in the url with `api/assemblies/d` leads me to an error 404.

My document is https://cad.onshape.com/documents/833f9762904995f22ff7dd9f/v/33f32ba5220d29406b8e50b0/e/83ac8f3e1b8a64890594cf88 

the configuration I want is "collisions", instead of "default". 

The following is my `config.json`
```
(venv) atticus@AJR-L5Pro:~/github/onshape_to_robot
$ cat catbot/config.json
{
        "documentId": "833f9762904995f22ff7dd9f",
        "outputFormat": "urdf",
        "versionId": "33f32ba5220d29406b8e50b0",
        "addDummyBaseLink": "true",
        "mergeSTLs": "all",
        "assemblyName": "robot_assm",
        "robotName": "catbot",
        "configuration": "collisions"
}

```

and this is the output of running the program. Any help on what I could change would be much appreciated.

```
(venv) atticus@AJR-L5Pro:~/github/onshape_to_robot
$ onshape-to-robot catbot/
pybullet build time: Jun 28 2023 20:16:25
* Checking OpenSCAD presence...
* Checking MeshLab presence...

* Using configuration version ID 33f32ba5220d29406b8e50b0 ...

* Retrieving elements in the document, searching for the assembly...
+ Found assembly, id: 83ac8f3e1b8a64890594cf88, name: "robot_assm"

* Retrieving assembly "robot_assm" with id 83ac8f3e1b8a64890594cf88

* Getting assembly features, scanning for DOFs...
WARNING: joint dof_front_left_wheel of type revolute has no limits
+ Found DOF: front_left_wheel (revolute)
WARNING: joint dof_front_right_wheel of type revolute has no limits
+ Found DOF: front_right_wheel (revolute)
WARNING: joint dof_rear_left_wheel of type revolute has no limits
+ Found DOF: rear_left_wheel (revolute)
WARNING: joint dof_rear_right_wheel of type revolute has no limits
+ Found DOF: rear_right_wheel (revolute)
* Found total 4 DOFs

* Building robot tree
* Trunk is still_chassis_assm <1>
* Adding top-level instance [still_chassis_assm <1>]
+ Adding part aluminum chassis plate <1> (configuration: List_F8qh5rve3N73aw=Default)
+ Adding part M3-5-60 spacer <2>
+ Adding part MD520z30_12v motor <4> (configuration: List_szyqBDS7R4nJ48=Default)
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M3X6-6  H type-N <11>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M3X6-6  H type-N <15>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M3X8-8  H type-N <9>
+ Adding part single coil lock washers light type gb_GB_FASTENER_WASHER_SSWL 3 <10>
+ Adding part single coil lock washers light type gb_GB_FASTENER_WASHER_SSWL 3 <17>
+ Adding part single coil lock washers light type gb_GB_FASTENER_WASHER_SSWL 3 <17>
+ Adding part single coil lock washers light type gb_GB_FASTENER_WASHER_SSWL 3 <19>
+ Adding part single coil lock washers light type gb_GB_FASTENER_WASHER_SSWL 3 <13>
+ Adding part M3-5-60 spacer <4>
+ Adding part aluminum chassis plate <2>
+ Adding part single coil lock washers light type gb_GB_FASTENER_WASHER_SSWL 3 <20>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M3X6-6  H type-N <15>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M3X6-6  H type-N <14>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M3X8-8  H type-N <10>
[... 113 lines cut ...]
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <3>
+ Adding part 6mm hex coupler <4>
+ Adding part 65mm wheel with high friction rubber tire <4> (configuration: List_cOWKJ6AUvxgRs2=Default)
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <7>
* Adding top-level instance [wheel_assm <2>]
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <3>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <11>
+ Adding part 6mm hex coupler <4>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <7>
+ Adding part 65mm wheel with high friction rubber tire <4> (configuration: List_cOWKJ6AUvxgRs2=Default)
* Adding top-level instance [wheel_assm <3>]
+ Adding part 6mm hex coupler <4>
+ Adding part 65mm wheel with high friction rubber tire <4> (configuration: List_cOWKJ6AUvxgRs2=Default)
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <7>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <3>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <11>
* Adding top-level instance [wheel_assm <4>]
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <3>
+ Adding part 6mm hex coupler <4>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <11>
+ Adding part cross recessed pan head screws gb_GB_CROSS_SCREWS_TYPE1 M4X8-8  H type-N <7>
+ Adding part 65mm wheel with high friction rubber tire <4> (configuration: List_cOWKJ6AUvxgRs2=Default)

* Writing URDF file
```

atticusrussell, 2023-07-04:
Unsure if I need to do anything with "add configuration input" for "list" or "configuration variable" - neither of which I currently have. 
[image]

Gregwar, 2025-02-20:
Hello,
This was improved on the latest version, you might be able to set a configuration like `Configuration=MyChoice` now

See:
https://onshape-to-robot.readthedocs.io/en/latest/config.html#configuration-default-default

Do not hesitate to re-open the issue if it doesn't work

closed 2025-02-20 by Gregwar