Rhoban/onshape-to-robot #105: configurations mess up filenames

atticusrussell, 2023-07-06:
The presence of configurations in an Onshape document results in a bunch of internal API rubbish being added to the names of the created files. This makes the URDF pretty gross also.
The way that configurations are handled in general is presently a bit broken - this is just one way the issue is manifesting itself.
```
(venv) atticus@AJR-L5Pro:~/github/onshape-to-robot/test  (master) 
$ onshape-to-robot catbot/
pybullet build time: Jun 28 2023 20:16:25
* Checking OpenSCAD presence...
* Checking MeshLab presence...

* Retrieving workspace ID ...
+ Using workspace id: 38852c293063cfa13b203677

* Retrieving elements in the document, searching for the assembly...
+ Found assembly, id: 83ac8f3e1b8a64890594cf88, name: "robot_assm"

* Retrieving assembly "robot_assm" with id 83ac8f3e1b8a64890594cf88

* Getting assembly features, scanning for DOFs...
WARNING: joint dof_front_left_wheel_speed_inv of type revolute has no limits 
+ Found DOF: front_left_wheel_speed (revolute)
WARNING: joint dof_front_right_wheel_speed of type revolute has no limits 
+ Found DOF: front_right_wheel_speed (revolute)
WARNING: joint dof_rear_left_wheel_speed_inv of type revolute has no limits 
+ Found DOF: rear_left_wheel_speed (revolute)
WARNING: joint dof_rear_right_wheel_speed of type revolute has no limits 
+ Found DOF: rear_right_wheel_speed (revolute)
* Found total 4 DOFs

* Building robot tree
* Trunk is wheel_assm <1>
* Adding top-level instance [wheel_assm <1>]
+ Adding part hex_coupler_6mm <4> (configuration: List_mX433oknDkb8hl=Default)
+ Adding part wheel <4> (configuration: List_cOWKJ6AUvxgRs2=Default)
* Adding top-level instance [still_chassis_assm <1>]
+ Adding part chassis_plate <1> (configuration: List_F8qh5rve3N73aw=Default)
+ Adding part spacer_M3-5-60 <2>
+ Adding part motor_MD520z30_12v <4> (configuration: List_szyqBDS7R4nJ48=Default)
+ Adding part spacer_M3-5-60 <4>
+ Adding part chassis_plate <2>
+ Adding part motor_bracket_MD520 <1> (configuration: List_E8lNvDhhLuVtnq=Default)
+ Adding part spacer_M3-5-60 <5>
+ Adding part motor_bracket_MD520 <1> (configuration: List_E8lNvDhhLuVtnq=Default)
+ Adding part motor_MD520z30_12v <4> (configuration: List_szyqBDS7R4nJ48=Default)
+ Adding part motor_MD520z30_12v <4> (configuration: List_szyqBDS7R4nJ48=Default)
+ Adding part motor_bracket_MD520 <1> (configuration: List_E8lNvDhhLuVtnq=Default)
+ Adding part spacer_M3-5-60 <6>
+ Adding part motor_MD520z30_12v <4> (configuration: List_szyqBDS7R4nJ48=Default)
+ Adding part spacer_M3-5-60 <3>
+ Adding part motor_bracket_MD520 <1> (configuration: List_E8lNvDhhLuVtnq=Default)
+ Adding part spacer_M3-5-60 <1>
* Adding top-level instance [wheel_assm <2>]
+ Adding part hex_coupler_6mm <4> (configuration: List_mX433oknDkb8hl=Default)
+ Adding part wheel <4> (configuration: List_cOWKJ6AUvxgRs2=Default)
* Adding top-level instance [wheel_assm <3>]
+ Adding part hex_coupler_6mm <4> (configuration: List_mX433oknDkb8hl=Default)
+ Adding part wheel <4> (configuration: List_cOWKJ6AUvxgRs2=Default)
* Adding top-level instance [wheel_assm <4>]
+ Adding part hex_coupler_6mm <4> (configuration: List_mX433oknDkb8hl=Default)
+ Adding part wheel <4> (configuration: List_cOWKJ6AUvxgRs2=Default)

* Writing URDF file
(venv) atticus@AJR-L5Pro:~/github/onshape-to-robot/test  (master) 
```
```
$ ls catbot/
chassis_plate__list_f8qh5rve3n73aw_default.part        motor_bracket_md520__list_e8lnvdhhluvtnq_default.stl
chassis_plate__list_f8qh5rve3n73aw_default.stl         motor_md520z30_12v__list_szyqbds7r4nj48_default.part
chassis_plate.part                                     motor_md520z30_12v__list_szyqbds7r4nj48_default.stl
chassis_plate.stl                                      robot.urdf
config.json                                            spacer_m3-5-60.part
hex_coupler_6mm__list_mx433okndkb8hl_default.part      spacer_m3-5-60.stl
hex_coupler_6mm__list_mx433okndkb8hl_default.stl       wheel__list_cowkj6auvxgrs2_default.part
motor_bracket_md520__list_e8lnvdhhluvtnq_default.part  wheel__list_cowkj6auvxgrs2_default.stl
```
```
(venv) atticus@AJR-L5Pro:~/github/onshape-to-robot/test  (master) 
$ cat catbot/config.json 
{
        "documentId": "833f9762904995f22ff7dd9f",
        "outputFormat": "urdf",
        "assemblyName": "robot_assm",
        "robotName": "catbot",
        "useScads": true
}
        
```

atticusrussell, 2023-08-31:
More info on this issue is also in #103. 

@Gregwar do you have any ideas about how best to tackle this in the code?

I'd love to have a simple solution from the side of config.json where the "list_XXXXXXX" does not need to be specified, but is found internal to the program by specifying the plaintext name of the configuration in config.json.

From playing with the code a bit in the debugger, I don't think it's a super simple fix, but would love the perspective of others.

Gregwar, 2025-02-20:
Hello,

There was many improvements in the codebase that should considerably improve the readability of produced URDF
Can you give a try to the latest release and give feedbacks?

closed 2025-04-14 by Gregwar