Rhoban/onshape-to-robot #109: create links made of multiple parts in longer open kinematic chains

Niklas-Mulsow, 2023-09-01:
Hello, 

I tried to make my first steps and models with your tool this week. First let my say thanks you for made it. 

I noticed, that it is not possible to connect the single parts to one link and then put another joint on it. So i had to put all parts for each links in a sub-assembly to built longer chains of joints and links. 

In the onshape to robot tutorial video, you did that also in that way, but i thought you not mentioned explicit that all parts had to be in subassemblies.

Anyway, do you plan to change that?

Gregwar, 2023-09-01:
Hello,

Actually, this might be possible, but this requires analyzing the constraints from the assembly to deduce that a given part is belonging to a specific link. 

I believe it could quickly result in hard-to-debug export problems, that is why I proposed to create a sub-assembly stage to have some hierarchy in the main assembly.

But yes, there is no theorical/technical limitation here, just some more software engineering

Niklas-Mulsow, 2023-09-05:
Hi, 

okay i can understand that. But then it is not just proposing. It is a must, to create subassemblies stages to make the export right. ;=)

Best
Niklas