Rhoban/onshape-to-robot #113: Unable to convert all of parts

drmwnrafi, 2023-11-11:
I've renamed the revolute joint to "dof_" but it only shows 1 part
[image]
This all parts robot preview on onshape:
[image]
Output :
[image]

This is the terminal log:
[image]

Gregwar, 2023-11-11:
Hello,

Can you share the output as a zip file?

From what I see you appear to have many parts in your top level assembly, I
would recommend to pre assemble the robot parts as sub assemblies

Also, can you see the joints in the produced URDF file?

drmwnrafi, 2023-11-11:
> semble the robot parts as sub assemblies

Hello,
This is the file : 
[outputs.zip](https://github.com/Rhoban/onshape-to-robot/files/13324921/outputs.zip)

Gregwar, 2023-11-11:
Indeed only one joint is appearing in the URDF itself

Did you click [children first](https://onshape-to-robot.readthedocs.io/en/latest/design.html#specifying-degrees-of-freedom) when creating DOFs?

drmwnrafi, 2023-11-11:
> Indeed only one joint is appearing in the URDF itself
> 
> Did you click [children first](https://onshape-to-robot.readthedocs.io/en/latest/design.html#specifying-degrees-of-freedom) when creating DOFs?

Yes, i clicked child first.
The parts for the revolute joint come from different documents, is that the problem?

Gregwar, 2023-11-11:
From what I see in your screenshot, I think the problem is that you dont
have subassemblies for the rigid parts of your robot

You cant have complex relations in your top level assembly, try having your
dofs only here

This is a current limitation of onshape to robot

Gregwar, 2025-02-26:
Hello,

I will close this because of lack of activity 
Do not hesitate to re-open if you have any updates

closed 2025-02-26 by Gregwar