Rhoban/onshape-to-robot #116: Retrieval of current joint states

Gregwar, 2024-01-18:
Onshape-to-robot don't retrieve the current joint states, i.e f OnShape joint states are different from the "zero" position (around which the limits are defined). As a result, the joint limits are supposed to be around the configuration of the robot during the import, which might be inconsistent with OnShape joint limits.

---

The current joint states might be available through the following endpoints:

```
/assemblies/d/{did}/{wvm}/{wvmid}/e/{eid}/matevalues
```

But it is not. This is a known bug and a support ticked is opened.

closed 2025-02-20 by Gregwar

hshi74, 2025-05-13:
Hi,  just came across this issue. Is it still a known bug?

Gregwar, 2025-05-13:
Yes, mate values are now retrieved and used as offset to shift the joint limits

Do you experience unwanted behaviour ?

hshi74, 2025-05-13:
I can confirm that the joint limits are correctly shifted, but should the "zero point" be shifted as well? This sounds like a design decision though...

Here are my test results. I'm testing the knee joint shown in the screenshots:

[image]

[image]

The joint limits are correct, but the zero point is the same as the current joint state in onshape.

Gregwar, 2025-05-13:
This is indeed a design choice. The argument is that Onshape zero if hardly configurable anyway, it roughly depends on the configuration of the mates when you created the relation.

Our practice is to have "lock"/"reset" constraints, which you can group, and unsupress/re-supress them to reset the robot in Onshape:

https://github.com/user-attachments/assets/e9471f35-ba54-4eb7-9f49-8b051826c6cf

In our case, they are all "parallel" constraints, resetting the dofs to a position that is the "zero" from the point of view of how our actual servos are configured

With tat, just make sure you reset the robot before running the export. Anyway, I also highly recommend making a version you are happy with, so that you can re-run the export while still tweaking the assembly. 

Hope this helps

hshi74, 2025-05-13:
Yes, that makes sense. Thanks for the detailed explanation! I use named positions to reset all the joints to 0 before exporting.

Gregwar, 2025-05-13:
I guess that is a good option as well. I didn't investigate, but named positions might be accessible through API, could be useful to recover some particular configurations as well

hshi74, 2025-05-13:
Yes, that would be super useful. On hindsight, named positions are exactly like [keyframes in MJCF](https://mujoco.readthedocs.io/en/stable/XMLreference.html#keyframe). It would be very cool if we can export named positions directly into keyframes in Mujoco. Looks like named positions are accessible through `/assemblies/d/{did}/{wvm}/{wvmid}/e/{eid}/namedpositions`

Gregwar, 2025-05-13:
This endpoint is giving the list of named positions indeed, but it looks like we can't do much else with it

[image]

hshi74, 2025-05-13:
I see, this is unfortunate.