Rhoban/onshape-to-robot #118: Add centimeter as an acceptable length unit and give error message more cotext

RrezonPllana, 2024-03-25:
### Motivation
I was trying to convert a robot of mine that I created in OnShape using `onshape-to-robot <my-robot>` but kept getting an error message `Unknown unit: cm` at first I really didn't understand what it meant until I spend couple of hours reading the whole onshape-to-robot repository, and then I saw in `readExpression()` method that the error was saying that my OnShape workspace length unit is centimeter something that is not supported by the project. So I tried to open a PR about this problem but I couldn't since I needed permission to write in the repository. The fix is below if someone wants to use it locally.

### Fix
```Diff
--- a/onshape_to_robot/features.py
+++ b/onshape_to_robot/features.py
@@ -49,11 +49,14 @@ def readExpression(expression):
             val = parts[0]
         return float(parts[0])
     elif parts[1] == 'mm':
-        return float(parts[0])/1000.0
+        return float(parts[0]) / 1000.0
+    elif parts[1] == 'cm':
+        return float(parts[0]) / 100.0
     elif parts[1] == 'm':
         return float(parts[0])
     else:
-        print(Fore.RED + 'Unknown unit: '+parts[1] + Style.RESET_ALL)
+        print(Fore.RED + 'Unknown unit: ' + parts[1] + '. Acceptable workspace units for angle are: radians or degrees, and for length '
+                                                       'are: millimeter, centimeter, or meter.' + Style.RESET_ALL)
         exit()

```

javiCuara, 2024-04-15:
I am having a similar issue with:
"Unknown unit: in " 
My project and account settings are all set to meter ...

Gregwar, 2024-04-18:
I think that some unit can be enforcedly defined to inches somewhere (maybe in some part you imported for e.g.)
We could support them though

Gregwar, 2024-04-18:
I added `cm` and `in` in `0.3.25` that I just pushed on pip, can you give it a try @javiCuara ?

closed 2025-02-20 by Gregwar