Rhoban/onshape-to-robot #122: When using parameters linked to variable studios - it will find the strings of the variable in the array, but not the value linked to it

minched, 2024-06-04:
Goodday,

I tried to link some parameters to variable studios. This way multiple document parameters are linked together/ the same. But also to make it easier to load and export these in and from Onshape. 

However the value is not inserted in the paramater, but only the string of the variable referring to is:

[image]

Leading to a crash becaue of unknown value:
[image]

I already tried configurations (https://github.com/Rhoban/onshape-to-robot/issues/103) But this worked only as well with numeric values and not with variables

Would it be possible to solve this - perhaps the same as with the configurations?

Thanks in advance for thinking with me.

Gregwar, 2025-02-26:
This should be handled in `1.3.3`, where variable values are retrieved if needed
Can you confirm it works ?

Gregwar, 2025-04-14:
Hello,
I will close this for now, do not hesitate to re-open if you want to provide more details

closed 2025-04-14 by Gregwar