Rhoban/onshape-to-robot #123: Unable to Locate Resources from Generated SDF File

EmDash00, 2024-06-05:
I've been trying to import this model for use in Gazebo Citadel (I believe) using Gazebo Sim v6 without much success.

CAD Model: https://cad.onshape.com/documents/1c63d90d811cafbc229b7bb7/w/839ee83dfd55d3b59e376792/e/b654b6774d2830de7cf6943b

I'm using the following config.json

```
{
    "documentId": "1c63d90d811cafbc229b7bb7",
    "outputFormat": "sdf",
}
```

```
❯ poetry run onshape-to-robot robot-project
pybullet build time: Nov 28 2023 23:45:17
* Checking OpenSCAD presence...
OpenSCAD version 2021.01
* Checking MeshLab presence...

* Retrieving workspace ID ...
+ Using workspace id: 839ee83dfd55d3b59e376792

* Retrieving elements in the document, searching for the assembly...
+ Found assembly, id: b654b6774d2830de7cf6943b, name: "robot"

* Retrieving assembly "robot" with id b654b6774d2830de7cf6943b

* Getting assembly features, scanning for DOFs...
+ Found DOF: d1 (prismatic)[-1.0: 1.0]
WARNING: joint dof_theta2 of type revolute has no limits
+ Found DOF: theta2 (revolute)
+ Found DOF: theta3 (revolute)[-3.142: 0.0]
+ Found DOF: theta4 (revolute)[-1.571: 1.571]
+ Found DOF: theta5 (revolute)[-1.571: 1.571]
WARNING: joint dof_theta6 of type revolute has no limits
+ Found DOF: theta6 (revolute)
* Found total 6 DOFs

* Building robot tree
* Trunk is slider_base <1>
* Adding top-level instance [slider_base <1>]
+ Adding part slider_base <1>
* Adding top-level instance [base <1>]
+ Adding part base <1>
* Adding top-level instance [joint connector <1>]
+ Adding part joint connector <1>
* Adding top-level instance [joint <1>]
+ Adding part joint <1>
* Adding top-level instance [joint <2>]
+ Adding part joint <2>
* Adding top-level instance [joint connector <3>]
+ Adding part joint connector <3>
* Adding top-level instance [joint end <4>]
+ Adding part joint end <4>

* Writing SDF file
```

When I try to import the resulting SDF file:

```
root@EmDashXPS9530:~/home/Documents/School/2024-sp/EE543/project/robot-project# Warning [Utils.cc:129] [/sdf/model[@name="onshape"]/link[@name="slider_base"]/inertial/pose[@relative_to="slider_base_frame"]:robot.sdf:L19]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="onshape"]/link[@name="base"]/inertial/pose[@relative_to="base_frame"]:robot.sdf:L41]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="onshape"]/link[@name="joint_connector"]/inertial/pose[@relative_to="joint_connector_frame"]:robot.sdf:L63]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="onshape"]/link[@name="joint"]/inertial/pose[@relative_to="joint_frame"]:robot.sdf:L85]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="onshape"]/link[@name="joint_2"]/inertial/pose[@relative_to="joint_2_frame"]:robot.sdf:L107]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="onshape"]/link[@name="joint_connector_2"]/inertial/pose[@relative_to="joint_connector_2_frame"]:robot.sdf:L129]: XML Attribute[relative_to] in element[pose] not defined in SDF.
Warning [Utils.cc:129] [/sdf/model[@name="onshape"]/link[@name="joint_end"]/inertial/pose[@relative_to="joint_end_frame"]:robot.sdf:L151]: XML Attribute[relative_to] in element[pose] not defined in SDF.
[Err] [SystemPaths.cc:378] Unable to find file with URI [file://robot-project/slider_base.stl]
[Err] [SystemPaths.cc:473] Could not resolve file [file://robot-project/slider_base.stl]
[Err] [MeshManager.cc:173] Unable to find file[file://robot-project/slider_base.stl]
[Err] [SystemPaths.cc:378] Unable to find file with URI [file://robot-project/base.stl]
[Err] [SystemPaths.cc:473] Could not resolve file [file://robot-project/base.stl]
[Err] [MeshManager.cc:173] Unable to find file[file://robot-project/base.stl]
[Err] [SystemPaths.cc:378] Unable to find file with URI [file://robot-project/joint_connector.stl]
[Err] [SystemPaths.cc:473] Could not resolve file [file://robot-project/joint_connector.stl]
[Err] [MeshManager.cc:173] Unable to find file[file://robot-project/joint_connector.stl]
[Err] [SystemPaths.cc:378] Unable to find file with URI [file://robot-project/joint.stl]
[Err] [SystemPaths.cc:473] Could not resolve file [file://robot-project/joint.stl]
[Err] [MeshManager.cc:173] Unable to find file[file://robot-project/joint.stl]
[Err] [SystemPaths.cc:378] Unable to find file with URI [file://robot-project/joint.stl]
[Err] [SystemPaths.cc:473] Could not resolve file [file://robot-project/joint.stl]
[Err] [MeshManager.cc:173] Unable to find file[file://robot-project/joint.stl]
[Err] [SystemPaths.cc:378] Unable to find file with URI [file://robot-project/joint_connector.stl]
[Err] [SystemPaths.cc:473] Could not resolve file [file://robot-project/joint_connector.stl]
[Err] [MeshManager.cc:173] Unable to find file[file://robot-project/joint_connector.stl]
[Err] [SystemPaths.cc:378] Unable to find file with URI [file://robot-project/joint_end.stl]
[Err] [SystemPaths.cc:473] Could not resolve file [file://robot-project/joint_end.stl]
[Err] [MeshManager.cc:173] Unable to find file[file://robot-project/joint_end.stl]
```

Though it does import when you do this.  I would appreciate if there were some SDF examples.

[image]

It's not super clear what the error messages indicate. Any tips also on programming the joints? I'm a bit of a new Gazebo user.

dragonblade316, 2024-09-27:
You need to set the [packageName attribute](https://onshape-to-robot.readthedocs.io/en/latest/config.html#packagename). If the files are in a subdirectory in the package then do this: "packagename/subfolder".

Probably a little late but I ran into this too so might as well put the answer here.

Gregwar, 2025-02-26:
In the last version, the model:// prefix is used
I am not that much of a SDF user, so if you have any feedbacks about that behaviour they are welcome

Gregwar, 2025-04-14:
Hello,
I will close this for now, do not hesitate to re-open if you want to provide more details

closed 2025-04-14 by Gregwar