Rhoban/onshape-to-robot #127: Standard parts exported in wrong size

pschmutz, 2024-10-05:
Using this example document https://cad.onshape.com/documents/82f6d830dd1d95c3f57f9bc3/w/9d326290f62aa21685913619/e/63495dc78a6cb9f6c9011616

With the following config

    {
        "documentId": "82f6d830dd1d95c3f57f9bc3",
        "outputFormat": "urdf",
    }

The model output looks like this
[image]

While the original model looks like this

[image]

The file names are correct though

[image]

Gregwar, 2025-02-20:
Hello, just tried this on the latest release and it appear to work better
Can you confirm it's ok for you ?

[image]

pschmutz, 2025-02-22:
Yes, the screw size is correct now. Thank you so much for fixing this and maintaining the tool. It is really useful :)

closed 2025-02-22 by pschmutz