Rhoban/onshape-to-robot #129: i am trying to create a urdf file but its showing some occurance issue

Creator-1705, 2024-10-29:
+ Using workspace id: 9f8cd5fea2c70356e51cacd8

* Retrieving elements in the document, searching for the assembly...
+ Found assembly, id: 50c24bab1c0738280bf27c29, name: "Assembly 1"

* Retrieving assembly "Assembly 1" with id 50c24bab1c0738280bf27c29
Traceback (most recent call last):
  File "/home/creator1705/.local/bin/onshape-to-robot", line 8, in <module>
    sys.exit(onshape_to_robot.main())
  File "/home/creator1705/.local/lib/python3.10/site-packages/onshape_to_robot/onshape_to_robot.py", line 17, in main
    from .load_robot import \
  File "/home/creator1705/.local/lib/python3.10/site-packages/onshape_to_robot/load_robot.py", line 139, in <module>
    for occurrence in root["occurrences"]:
KeyError: 'occurrences'

Gregwar, 2025-02-20:
Can you try again on the latest version and provide the output ?

Gregwar, 2025-04-14:
Hello,
I will close this for now, do not hesitate to re-open if you want to provide more details

closed 2025-04-14 by Gregwar