Rhoban/onshape-to-robot #140: Linear relation and Gear relation

Joro2220, 2025-02-27:
Hi Grégoire,

I haven't seen support for Linear mate relation that would make telescoping robot possible for example.
Maybe this could be done by naming the mate relation "mimic_previous" or "mimic_next" to indicate which joint is the driving joint. Then, the multiplier would be the "Relation ratio" and the sine of the multiplier would be the "Reverse direction" check box.

Is that a feature you were planning to implement?

Best regards!

Joël

Gregwar, 2025-02-27:
Hello,

Onshape supports Gear relations:

[image]

This is not (yet) retrieved by onshape-to-robot, but it is possible to implement

Gregwar, 2025-02-27:
Can you specify the export format you are using, and the tool you use that takes the mimic tags in account ?

I know I can test that with MuJoCo equality constraints, but it seems that pyBullet and Gazebo don't care much about `<mimic>` tags

Gregwar, 2025-02-27:
Just pushed `1.3.4` supporting gear relations. Do not hesitate to provide on this.

https://github.com/user-attachments/assets/0738a37b-1c27-46b1-abb1-b95958860bf0

https://github.com/user-attachments/assets/85b95fb7-d8a2-46a2-add7-4375837fcc20

Joro2220, 2025-03-04:
Hi Grégoire,

I tried the mate relation feature and it looks to be working great! The only thing is that I had a fraction entered in the relation ratio so an error occurred. With a decimal number it worked fine, but it would be nice to support basic mathematical operations.

Thx again!

Joël

Gregwar, 2025-03-04:
Better handling of Onshape expressions is definitely something I want to improve
Keep this open for now

Gregwar, 2025-04-14:
Hello,
Is this working as expected for you now ?

Gregwar, 2025-05-11:
I am closing this for now, do not hesitate to re-open if you encounter issue with this feature

closed 2025-05-11 by Gregwar