Rhoban/onshape-to-robot #141: urdf and mujoco generation issues

nfreq, 2025-02-28:
Before I start, fantastic tool!

For URDF Generation, the joint properties have a couple issues:

```
 <joint name="right_shoulder_pitch" type="revolute">
        <origin xyz="-0.029 -0.01025 0.00305001" rpy="-3.14159 8.17357e-20 0.0738744"/>
        <parent link="right_shoulder_pitch_motor"/>
        <child link="right_shoulder_yaw"/>
        <axis xyz="0 0 1"/>
        <limit effort="1" velocity="10" lower="-1.74533" upper="1.74533"/>
        <joint_properties friction="{'max_effort': 1.0, 'max_velocity': 10.0, 'friction': 10.0, 'limits': [-1.7453293, 1.7453293]}"/>
    </joint>
```
[config.json](https://github.com/user-attachments/files/19024117/config.json)
[robot_urdf.txt](https://github.com/user-attachments/files/19024136/robot_urdf.txt)

For MJCF Generation the joint properties don't appear to be used. Also, I have. "Mate Connection" named frame_imu which I assumed would show up as a site in the configuration.

[config.json](https://github.com/user-attachments/files/19024176/config.json)
[robot_xml.txt](https://github.com/user-attachments/files/19024181/robot_xml.txt)

Am I missing anything?

Gregwar, 2025-02-28:
Hello,

Thank you for reporting, many refactor was performed and there are likely still a few typos/glitches remaining

About URDF export: it should be fixed in `1.3.8`

About MuJoCo: the parameters in joint_properties have different names, in order to stick with MuJoCo vocabulary. I noticed documentation still had some old examples, just updated them:
https://onshape-to-robot.readthedocs.io/en/latest/exporter_mujoco.html

About the frame: is there a way I can see the OnShape assembly ?

Gregwar, 2025-02-28:
About the frame, if you want you can create a simple document with two bodies and a frame, see if you can reproduce your issue there, and share this simple document with me

Is your frame a single mate connector or is it a mate connection between two parts?

nfreq, 2025-02-28:
Thanks for the quick response.

Yes, feel free to inspect directly. The url is in the config.json file.

nfreq, 2025-02-28:
> Ok, what version of Onshape to robot are you using?

Great question. I pip installed it at approx 8-9pm EST tonight. I am afk but will install the latest version in the morning and repeat tests and report back. Will also create simple example of frame/site and see what happens too.

Gregwar, 2025-02-28:
Here is a slightly modified version:
https://cad.onshape.com/documents/4605e3fe923b4af92600a00f/w/0fd698dc31859f1369f208e3/e/143f772f32ad3617cf91e6ec

Resulting in:

https://github.com/user-attachments/assets/bf089ea7-41e0-4896-a852-bb3292a16521

I mostly renamed all mates that are fixing mates with `fix_` prefix. Doing so gets the bodies merged upstream and is better than using fixed DOFs.

I would suggest that you make subassemblies to fix parts together to mitigate the number of links in the top-level assembly. Consider ignoring non-relevant parts for collisions.

Also, if the top-level assembly is done in a separate document, it will allow caching.

nfreq, 2025-02-28:
Thank you for looking at this, I appreciate your time!

It appears your project is not public, can not access to review.

I was running 1.3.3.  Upgrading to 1.3.8 does indeed fix the joint properties issue. I also not see a site for my imu!  I changed the names to fix_ for all of my fixed mates.  Understood on your suggestions for improvements on cad side.

Did you make any modifications to the config.json that gave you the above result?  I'd like to compare onshape projects when you get a chance.

Gregwar, 2025-02-28:
Made my copy public
I didn't change the `config.json`

nfreq, 2025-02-28:
Thank you very much for the ultra fast response and support.   This tool is awesome.

closed 2025-02-28 by nfreq