Rhoban/onshape-to-robot #147: URDF Export Doesn't Contain Base Link Mass and Inertia Properties

kyllthan8, 2025-03-31:
The values in the URDF file are 0 for mass and inertia. I am using onshape-to-robot version 1.4.7

Gregwar, 2025-03-31:
Hello
Can you share your Onshape model and config.json ?

Gregwar, 2025-03-31:
Also, is your bass link fixed?

Gregwar, 2025-03-31:
(If you have fixed your base in Onshape it will be considered as a fixed robot
Inertia is omitted in that case)

kyllthan8, 2025-03-31:
Ah ok, my bad. I inadvertently had the part fixed. I didn't realize that it ignored the base link properties when it was fixed. I unfixed it and it does not have any issues now

closed 2025-03-31 by kyllthan8

Gregwar, 2025-03-31:
This behavior is because of pybullet, where the convention is to consider world-fixed the bodies with 0 mass
However, maybe it should be changed, since the presence of a floating base should not be related to the absence of inertial infos

kyllthan8, 2025-03-31:
Yeah it causes errors when importing into other simulation tools (mujoco specifically)

Gregwar, 2025-03-31:
After a second thought, I updated the behaviour so that inertial data are not removed from fixed joints
I think this is a bullet-specific thing, and not a wide convention for URDF users

I added an option to recover this behaviour
https://onshape-to-robot.readthedocs.io/en/latest/exporter_urdf.html#set-zero-mass-to-fixed-default-false

kyllthan8, 2025-03-31:
Awesome, thanks!