Rhoban/onshape-to-robot #148: [Feature Request] Adding registration method for custom preprocessors

T-K-233, 2025-04-03:
Version 1.x release is awesome! I really like the clean separation of the core functionality and the preprocessors as extensions. 

I wonder if there is any plan on adding support to allow user to register a custom preprocessor class externally, so user can use the upstream installation of onshape-to-robot.

Thanks for creating this tool.

Gregwar, 2025-04-03:
Hello,

This indeed would be possible.  Registering processors could be an option of `config.json`:

```json
"processors": [
    "my_module.my_processor:MyProcessor"
]
```

Importing automatically from the string can be done, even if it requires `exec()`.

The problem here is the processors *order.* For example, you want SCAD to be processed before the parts are merged. You also want the collision meshes to be removed after SCAD are possibly processed to keep pure shapes only etc.

So one would need to specify how his processor would be ordered with respect to all other processors (or at least the one you use).

```js
// Adding a custom processor after default ones
"processors": [
    "defaults", // All default processors
    "my_module.my_processor:MyProcessor"
]
```

Alternatively, this would allow trickier use case like the following:

```js
// Adding a custom processor in the middle of the existing processors
"processors": [
    "ProcessorScad",
    "ProcessorMergeParts",
    "ProcessorSimplifySTLs",
    "ProcessorFixedLinks",
    "my_module.my_processor:MyProcessor"
    "ProcessorNoCollisionMeshes",
    "ProcessorCollisionAsVisual"
]
```

The existing processors list is quite easy to list in https://github.com/Rhoban/onshape-to-robot/blob/master/onshape_to_robot/processors.py

Gregwar, 2025-04-08:
Hello again @T-K-233 : any second thoughts, or does my proposition sounds good for you ?

T-K-233, 2025-04-08:
Oh sorry I forgot to reply. 

Yeah this sounds good. For ordering, I prefer the second approach, where each processor will be listed out individually and explicitly according to their execution order. `defaults` might create some ambiguity in case when new processors are added to the standard set and messes up the ordering.

Gregwar, 2025-04-09:
I just added the feature. Documentation is here:
https://onshape-to-robot.readthedocs.io/en/latest/custom_processors.html

It is also live in `1.5.7`. Do not hesitate to re-open if any rework is needed

closed 2025-04-09 by Gregwar