Rhoban/onshape-to-robot #149: Unable to call onshape-to-robot command

d1ssapointed, 2025-04-10:
This is my first time using this program; whenever I use the command onshape-to-robot as per the documentation, it returns an error. This happens before and after exporting the API key & using mkdir (not sure if that's what they're called, sorry.)

$ onshape-to-robot my-robot
Traceback (most recent call last):
  File "C:\Users\#####\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\#####\AppData\Local\Programs\Python\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\#####\AppData\Local\Programs\Python\Python39\Scripts\onshape-to-robot.exe\__main__.py", line 7, in <module>
  File "C:\Users\#####\AppData\Local\Programs\Python\Python39\lib\site-packages\onshape_to_robot\export.py", line 8, in main
    from .robot_builder import RobotBuilder
  File "C:\Users\#####\AppData\Local\Programs\Python\Python39\lib\site-packages\onshape_to_robot\robot_builder.py", line 8, in <module>
    from .assembly import Assembly
  File "C:\Users\#####\AppData\Local\Programs\Python\Python39\lib\site-packages\onshape_to_robot\assembly.py", line 6, in <module>
    from .robot import Joint
  File "C:\Users\#####\AppData\Local\Programs\Python\Python39\lib\site-packages\onshape_to_robot\robot.py", line 93, in <module>
    class Joint:
  File "C:\Users\#####\AppData\Local\Programs\Python\Python39\lib\site-packages\onshape_to_robot\robot.py", line 113, in Joint
    limits: tuple[float, float] | None = None,
TypeError: unsupported operand type(s) for |: 'types.GenericAlias' and 'NoneType'

My running theory is that somehow I installed it wrong (I used pip install onshape-to-robot). Restarting command prompt doesn't help, and I haven't seen this issue anywhere else. Any help would be much appreciated.

Gregwar, 2025-04-11:
Do you have access to Python 3.10 or higher, and can you confirm it works?
I could update type hinting to recover 3.9 compatibility, but it's really close to being end-of-life

d1ssapointed, 2025-04-11:
Hi, just downgraded to 3.10.10, and it works fine. Thank you so much. I was using 3.19.3 before if it means anything.

Gregwar, 2025-04-11:
I believe you upgraded instead of downgrading and mean 3.9.3 instead of 3.19.3. The latest released version version is 3.13.3 (see: https://www.python.org/doc/versions/)

d1ssapointed, 2025-04-11:
You're right, sorry about that; I was using 3.13.3, just checked lol. Thank you nonetheless

closed 2025-04-11 by d1ssapointed