Rhoban/onshape-to-robot #150: onshape-to-robot my-robot doesn't create xml files

questions-git, 2025-04-10:
When I use the command onshape-to-robot my-robot it creates the urdf format and assets but not the robot.xml, scene.xml files that's why I can't simulate on Mujoco(onshape-to-robot-mujoco my-robot). How can I solve this problem?

Gregwar, 2025-04-10:
You need to specify the `output_format` entry in your `config.json`

For example, for URDF:
https://github.com/Rhoban/onshape-to-robot-examples/blob/master/2wheels_urdf/config.json

And for MuJoCo:
https://github.com/Rhoban/onshape-to-robot-examples/blob/master/2wheels_mujoco/config.json

Gregwar, 2025-04-10:
(If you want both output, you currently need to run two exports, since the options are generally slightly differents)

Gregwar, 2025-04-11:
Hello, did you manage to export your XML ?

Gregwar, 2025-05-11:
I am closing this for now, do not hesitate to re-open if you still have an issue

closed 2025-05-11 by Gregwar