Rhoban/onshape-to-robot #151: Getting an issue with typeerrors

CodeyVoiceDude, 2025-04-16:
Traceback (most recent call last):
  File "/home/funkyvoicedude/.local/bin/onshape-to-robot", line 5, in <module>
    from onshape_to_robot import export
  File "/home/funkyvoicedude/.local/lib/python3.8/site-packages/onshape_to_robot/export.py", line 5, in <module>
    from .robot_builder import RobotBuilder
  File "/home/funkyvoicedude/.local/lib/python3.8/site-packages/onshape_to_robot/robot_builder.py", line 6, in <module>
    from .assembly import Assembly, INSTANCE_ROOT
  File "/home/funkyvoicedude/.local/lib/python3.8/site-packages/onshape_to_robot/assembly.py", line 6, in <module>
    from .robot import Joint
  File "/home/funkyvoicedude/.local/lib/python3.8/site-packages/onshape_to_robot/robot.py", line 5, in <module>
    class Part:
  File "/home/funkyvoicedude/.local/lib/python3.8/site-packages/onshape_to_robot/robot.py", line 18, in Part
    color: tuple | None = None,
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'

I believe I followed everything exactly? And am using python version 3.8

Gregwar, 2025-04-16:
Hello,

Indeed, some syntax used is not compatible with Python 3.8

Can you consider upgrading to Python 3.10 or higher?
Python 3.8 is end of life:  https://devguide.python.org/versions/

CodeyVoiceDude, 2025-04-16:
Unfortunately i'm not able to do that at this moment, but thank you for the response!

Gregwar, 2025-05-02:
Hello @CodeyVoiceDude 

If you update `onshape-to-robot` to `1.6.0` with:

```
pip install -U onshape-to-robot
```

You should have a version working with Python3.9. Do not hesitate to reopen if the issue is still there

closed 2025-05-02 by Gregwar