Rhoban/onshape-to-robot #155: joint_properties "limits" not factored for SDF

bpousett, 2025-05-19:
Using onshape-to-robot version 1.5.7 to generate SDF output (SDF version 1.7).

In the documentation, you can use the joint_properties tag to set joint limits for SDF: https://onshape-to-robot.readthedocs.io/en/latest/exporter_sdf.html

However, using this in the config.json file for SDF export places the joint position limits (lower and upper) outside of the limit element.
```
    <!-- Joint from link1 to link2 -->
    <joint name="joint2" type="revolute">
      <pose relative_to="link1">0.32 -1.10956e-14 0.0615 1.69034e-17 -1.85345e-17 -1.4399</pose>
      <parent>link1</parent>
      <child>link2</child>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <effort>50</effort>
          <velocity>10</velocity>
        </limit>
        <lower>0.262</lower>
        <upper>2.62</upper>
        <dynamics>
          <friction>0.1</friction>
          <damping>0.0</damping>
        </dynamics>
      </axis>
    </joint>
```

This results in a parsing error, since the lower and upper elements are children of the limit element according to the SDF documentation: http://sdformat.org/spec?ver=1.7&elem=joint. It results in a parsing error.

Gregwar, 2025-05-19:
Hello,
Thank you for reporting, should be fixed in `1.6.6`, can you confirm it's ok?

bpousett, 2025-05-19:
Confirmed that it has been fixed. Thank you very much.

closed 2025-05-19 by bpousett