Rhoban/onshape-to-robot #156: Using scad with mulitple shape instances leads to duplicate names in SDF

bpousett, 2025-05-20:
Python v3.10.12
onshape2robot v1.6.6

Here is an example project (pulling a trivial cube, publicly available): [test.zip](https://github.com/user-attachments/files/20331520/test.zip)

In this project I chose to use openscad to generate my collision geometry. In the case that there is more than one shape, for example:
```
translate([-30, -30, 0])
cube([60, 30, 25]);

translate([-30, 0, 0])
cube([60, 30, 25]);
```
then the resulting sdf file has duplicate names for collision geometry:
```
      <!-- Part cube -->
      <collision name="cube_collision_shapes_1">
        <pose relative_to="cube">-4.30829e-18 -0.015 0.025 0 -0 0</pose>
        <geometry>
          <box>
            <size>0.06 0.03 0.025</size>
          </box>
        </geometry>
      </collision>
      <collision name="cube_collision_shapes_1">
        <pose relative_to="cube">-4.30829e-18 0.015 0.025 0 -0 0</pose>
        <geometry>
          <box>
            <size>0.06 0.03 0.025</size>
          </box>
        </geometry>
      </collision>
```
The parser complains as a result of the number "_1" not incrementing. I have duplicated this bug on more complicated examples which require multiple collision geometry instances. This is not an issue in urdf, since the collision tags are not named.

Gregwar, 2025-05-20:
Indeed, sounds like a refactoring mistake, should be fixed now (pushed `1.6.7`)

bpousett, 2025-05-20:
Confirmed that it is fixed.