Rhoban/onshape-to-robot #157: inertial/pose[@relative_to=] not defined in SDF

bpousett, 2025-05-20:
Thank you for making such a fantastic tool :)

Python v3.10.12
onshape2robot v1.6.6

The SDF exporter generates files with the "pose" element inside the "inertial" element having the "relative_to" attribute as follows:
```
  <model name="test">
    <!-- Link cube -->
    <link name="cube">
      <inertial>
        <pose relative_to="cube">-2.35672e-18 -2.0357e-18 0.0125 0 0 0</pose>
        <mass>0.7065</mass>
        <inertia>
          <ixx>0.000248747</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>0.000248747</iyy>
          <iyz>0</iyz>
          <izz>0.0004239</izz>
        </inertia>
      </inertial>
```

Even though, according to the [documentation of SDFormat version 1.7](http://sdformat.org/spec?ver=1.7&elem=link), it says that this "relative_to" attribute should be supported, it is not:

Gazebo complains via a warning:
```
Warning [Utils.cc:132] [/sdf/model[@name="test"]/link[@name="cube"]/inertial/pose[@relative_to="cube"]:/home/brendan/tmp/onshape2robot/test/robot.sdf:L9]: XML Attribute[relative_to] in element[pose] not defined in SDF.
```

Drake parsing fails altogether:
```
RuntimeError: /home/brendan/tmp/onshape2robot/test/robot.sdf:9: error: At XML path /sdf/model[@name="test"]/link[@name="cube"]/inertial/pose[@relative_to="cube"]: XML Attribute[relative_to] in element[pose] not defined in SDF.
```

"relative_to" is supported in collision and visual. It is only in inertial that it is causing problems. Since the inertial seems to be always relative_to the parent link, which is default in SDF, one solution could just be to remove it.

Gregwar, 2025-05-20:
Thanks for reporting. Updated the behaviour in `1.6.8`, can you confirm it works for you ?

bpousett, 2025-05-20:
There is now some issue in add_inertial. Here is my trace when pulling the same 'cube' project:
```
* Checking OpenSCAD presence...
OpenSCAD version 2021.01
* Not workspace ID specified, retrieving the current workspace ...
+ Using workspace id: ec50d76ba50715b67efe4492

* Retrieving elements in the document, searching for the assembly...
* Retrieving assembly with id 44dcd892a2efd69e72bdacd6
* Found total 0 degrees of freedom
* Found 1 root nodes:
  - cube <1>

+ Adding part cube <1>
+ Parsing OpenSCAD files...
+ Removing collision meshes
ERROR: must be real number, not str
Traceback (most recent call last):

...

  File "/home/brendan/.virtualenvs/onshape2robot/lib/python3.10/site-packages/onshape_to_robot/exporter_sdf.py", line 85, in add_inertial
    '<pose>%g %g %g 0 0 0</pose>'
TypeError: must be real number, not str
```

Gregwar, 2025-05-20:
My bad,  can you update again ?

closed 2025-05-20 by bpousett