Rhoban/onshape-to-robot #163: Error 400 when trying to export my model.

DumbleDuck, 2025-06-30:
I am developing an AMR which I have assembled in OnShape. The thing is that previously, I have successfully used the plugin to convert the same assembly to MJCF & URDF. But after adding a few more parts, I am consistently getting ERROR 400. The error is something like: 

> https://cad.onshape.com/api/parts/d/DocID/m/someNumber/e/moreNumbers/partid/JF%2f/massproperties?configuration=default&useMassPropertyOverrides=True&linkDocumentId=DocID
! ERROR (400) while using Onshape API 

Things that changed since last successful export:

1. Created a BasePlate part studio file & added it to my robot assembly. Have tried removing it from assembly but the error persists.
2. Created a read-only link to be added in my Resume.
3. Created new API keys.

Gregwar, 2025-06-30:
I believe this is an issue with Onshape API, I opened a support ticket and will keep you on touch

Minimal example reproducing the issue:
https://cad.onshape.com/documents/aa0aedb1590c3b6396471e81/w/1c229b75ac41da7b56f19ac9/e/5ed75bdc58c0a88577ccf477

* `Part 1` has partid `JHD`
* `Part 17` has part `JI/`

Using Onshape Glassworks, the first works and the second triggers an error 400 just like onshape-to-robot

Ticket 384778 for reference

[image]

Gregwar, 2025-06-30:
This issue has been confirmed as a bug by Onshape team

DumbleDuck, 2025-06-30:
Thank you for such a prompt response. I would be happy to share my the link of my model. As for the workaround, I couldn't find any partId containing "/".

carltech22, 2025-07-03:
Did the 400 issue ever get resolved?
I am experiencing the same issue here even though I was able to export a similar assembly to URDF like last week before the onshape update. 
"! ERROR (400) while using Onshape API"

Gregwar, 2025-07-03:
I believe this is a regression bug in the Onshape API

As I mentioned above, I opened a support ticket with a minimal example and they acknowledged it as a bug which they are currently fixing 

Escaping / characters in partid was introduced in onshape-to-robot a while ago and I am pretty sure it used to work

carltech22, 2025-07-07:
Thank you for your quick reply. I was working with parts from a supplier that were in French. After doing some digging it seems like some characters in French were treated as escaping characters which would bring up the Error (400)

Gregwar, 2025-07-09:
Indeed, this bug was fixed by Onshape!
Thanks for reporting