Rhoban/onshape-to-robot #164: TypeError while importing

nameofuser1, 2025-07-06:
Trying to import my assembly and get the following error:

```
Traceback (most recent call last):
  File "/home/yuriy/.local/bin/onshape-to-robot", line 5, in <module>
    from onshape_to_robot import export
  File "/home/yuriy/.local/lib/python3.7/site-packages/onshape_to_robot/export.py", line 5, in <module>
    from .robot_builder import RobotBuilder
  File "/home/yuriy/.local/lib/python3.7/site-packages/onshape_to_robot/robot_builder.py", line 6, in <module>
    from .assembly import Assembly, INSTANCE_ROOT
  File "/home/yuriy/.local/lib/python3.7/site-packages/onshape_to_robot/assembly.py", line 6, in <module>
    from .robot import Joint
  File "/home/yuriy/.local/lib/python3.7/site-packages/onshape_to_robot/robot.py", line 5, in <module>
    class Part:
  File "/home/yuriy/.local/lib/python3.7/site-packages/onshape_to_robot/robot.py", line 19, in Part
    shapes: list[Shape] | None = None,
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'

```

Gregwar, 2025-07-06:
Hello,

It appear you don't have the latest `onshape-to-robot` version, can you update it with the following command ?

`pip install -U onshape-to-robot`

Gregwar, 2025-07-23:
I am closing this for inactivity, don't hesitate to reopen if you still have an issue

closed 2025-07-23 by Gregwar